DreamSourceLab/DSView #651: SPI decode data in one csv-file (MISO and MOSI)

LEH-Tek, 2023-06-20:
It would be great if it was possible to save all SPI decode data in one csv-file MISO and MOSI, It will be easier for me to analyze if I can get MISO and MOSI data in one csv-file today you can only save one file separately for MISO and MOSI in DSView I then need to merge the data in the same file afterwards.

[image]

dreamsource-tai, 2023-06-28:
@LEH-Tek 
We need to evaluate this feature. Thank you!

dreamsource-tai, 2023-07-03:
@LEH-Tek 
We have already implemented this feature. If your operating system is Linux, you can pull the code and compile it to experience it.

LEH-Tek, 2023-07-07:
Thanks I use Windows unfortunately but good that it is implemented already.