DreamSourceLab/DSView #654: Display data in real time mode

saviotta, 2023-06-23:
At the best of my knowledge there's no possibility of real time display, i.e. have on screen data traces meanwhile capture is ongoing, and this is a big issue. Presence of this feature is what makes some competitor a better choice.
Do you have any plan to have real time display of data?

dreamsource-tai, 2023-06-26:
@saviotta 
V1.3 version is currently being tested and has real-time refresh functionality.

radorodopski, 2023-07-07:
I second that request!
Could you please release a beta of V1.3 that we can test and provide further feedback? I am really missing this feature.

DreamSourceLab, 2023-07-09:
@radorodopski @saviotta 
You can download the beta version of V1.3 for windows using this link:
https://www.dreamsourcelab.com/software/DSView_v1.3.0-RC9_x64_setup.exe