DreamSourceLab/DSView #655: Crash DSView

juf00th, 2023-07-02:
Hello.
On update "Disable many warning messages" DSView crashed!

`DSView: ----------------- version: 1.3.0-RC9-----------------
DSView: Qt:5.15.8
DSView: 2023-07-02 10:16:40
DSView: GetAppDataDir:"/usr/local/share/DSView"
DSView: GetFirmwareDir:"/usr/local/share/DSView/res"
DSView: GetUserDataDir:"/home/art/.local/share/DreamSourceLab/DSView"
DSView: GetDecodeScriptDir:"/usr/local/share/libsigrokdecode4DSL/decoders"
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found new DSLogic device count: 0
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: Hotplug thread start!
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
Помилка адресування`

dreamsource-tai, 2023-07-03:
@juf00th 
What is your operation system? Is you code is the latest?

dreamsource-tai, 2023-07-03:
@juf00th 
We fixed the issue, please pull the code. Thank you!

closed 2023-07-07 by juf00th