DreamSourceLab/DSView #657: Feature request: Hide channels

mrehkopf, 2023-07-07:
Would be handy if channels could be made invisible. I work with parallel buses a lot (up to 24 bit) and I do not need to see all the single bit channels to take up window space. So it would be convenient if they can be hidden.

dreamsource-tai, 2023-07-07:
@mrehkopf 
Click the "Options" button on toolbar to open options dialog,  hide the channel by unchecking the checkbox.

mrehkopf, 2023-07-07:
In that case, the channel is disabled and not captured, and the protocol decoder gives an error message that the channel is not available. I need the channel to be captured, just not displayed.

dreamsource-tai, 2023-07-07:
@mrehkopf 
We need some time to implement the feature you mentioned. This is a function of virtual channels.

mrehkopf, 2023-07-07:
Thank you! Yes, hiding after capture is a workaround.