DreamSourceLab/DSView #661: Feature request: Manually enter frequency in external clock mode

mrehkopf, 2023-07-07:
When using external clock mode, the time measurements are incorrect because the (expected) frequency of the external clock cannot be entered. In that case it would be helpful if the frequency or period could be entered manually to get correct time readings.

Wassabi56, 2023-07-18:
Same issue here.
First, the external clock has to be 3.3V or higher, or you will never get a trigger.
Then, there is no selection in the GUI sampling rate drop-down for "External Clock".
The hardware samples at the frequency chosen in the drop-down and NOT at the frequency of the external clock.
External sampling clock simply does not work.

mrehkopf, 2023-07-18:
I only tested with a 5V clock (~21MHz) and that worked for me, sampling also took place at the rate of the external clock. I selected external clock via the Options menu. But timing displays were wrong because they use whatever clock is selected in the drop down list.

Wassabi56, 2023-07-18:
Thanks, I will try a 5V external clock.
With a 3.3V 768KHz clock, I captured for 2 seconds and the .csv file that I exported had 1M rows, which matches the 500KHz sampling that I had selected in the drop-down menu.

Did you check that the .csv file had for example ~2.1M samples if you captured for 100ms?

mrehkopf, 2025-01-15:
This has been a while... anyway, I just checked the CSV export and it contains around 2.5M samples with the 21MHz clock. I think this happens because the number of samples to capture, and the capture timestamps, are also calculated based on the configured sampling frequency, not real time.

mrehkopf, 2025-01-18:
Ended up doing it myself, in a way that can hopefully be useful.
#845