DreamSourceLab/DSView #666: More languages

juf00th, 2023-07-18:
Are there plans to add more languages?
I can help with the translation.
If there is a convenient file for translation.

dreamsource-tai, 2023-07-19:
@juf00th 
What language do you want to translate? Please take a look at the files in the directory "lang/en" and translate them into new files. Upload them and we will integrate them. At present, it is not possible for users to add their own language configuration. The next version may consider this feature. Under Linux, this directory is: "/usr/local/share/DSView/lang/en".

juf00th, 2023-07-19:
@dreamsource-tai 
> @juf00th What language do you want to translate?
In the Ukrainian language.
I was looking at the wrong files. As I translate, I will write. Thank you.

dreamsource-tai, 2023-07-21:
@juf00th 
Please refer to the direcotry "lang/cn" for more translation items.

juf00th, 2023-07-21:
@dreamsource-tai 
I understood that if there is no file, the "id" value is taken: from the file in the cn directory. OK.
Thank you.

juf00th, 2023-07-27:
@dreamsource-tai 
Translation files here:
https://webart4.me/_cloud/s/BQYHSx3QYtqRZgB

petrfirst4786, 2025-03-20:
> [@dreamsource-tai](https://github.com/dreamsource-tai) Translation files here: https://webart4.me/_cloud/s/BQYHSx3QYtqRZgB
> 
> "Українська" "ua"

Вітаю!. Посилання вже застаріле. Моє є змога оновити, або в особисті надіслати локалізацію?