DreamSourceLab/DSView #667: Packet error, "The received packet format is incorrect!"

kasra3422, 2023-07-24:
I recently encountered an issue with my C20P device, which had been working with version 1.2.2 on my Ubuntu system. However, upon updating to version 1.3.0, I encountered an error message stating, "The received packet format is incorrect!"

[image]

dreamsource-tai, 2023-07-25:
@kasra3422 
Does this error occur every time?

dreamsource-tai, 2023-07-25:
@kasra3422 
Try connecting to the USB port of the PC instead of the hub.

kasra3422, 2023-07-25:
> @kasra3422 Try connecting to the USB port of the PC instead of the hub.

Thank you for your assistance! I managed to fix the issue by directly connecting the C20P device to my PC. Surprisingly, after that initial connection, it started working flawlessly even when connected through the hub.

l0ud, 2023-07-25:
I have the same problem with DSLogic PLus, Manjaro Linux and 1.3.x version.
It happens every time. Usually I get "The received packet format is incorrect" and whole capture process is not responsive (I cannot even cancel it reliably).
I'm not using any hubs, analyzer is directly connected to my laptop.
Downgrading to 1.2.2 fixes the issue completely.

kasra3422, 2023-08-02:
It initially worked for a few days, but unfortunately, the same problem has resurfaced. Additionally, after attempting multiple connections and disconnections, it eventually starts working, although it still exhibits some issues with rubbish data in the signals, and the signals captured on the screen keep changing.

dreamsource-tai, 2023-08-08:
@kasra3422 
What is your operation system? If is linux, please help us to test the new version for this issue. The new version at the branch "release-1.3". Thank you!

markwu2001, 2023-08-08:
Hey! I had the exact same problem when installing v1.30 on macOS. "The received packet format is incorrect!" was what I saw too. I don't remember this happening so I downloaded the 1.2.2 release on GitHub and I am able to see normal measurements again. I have a M2 macbook.

OpusElectronics, 2023-10-12:
Same problem here, v1.3.0 on Arch Linux.
I get the same error every time a capture is done. The device is connected directly to the PC and is correctly recognized by the DSView app, LED becomes green.

If v1.2.2 is known to work, I'll try compiling it and see.