DreamSourceLab/DSView #669: Please add: Programmable Math Functions, a field where you can enter your formula.

cumtliang, 2023-07-27:
Math functions.  A-B A+B and so on is ok, but Better would be a field where you can enter your formula.

like 
[image]

dreamsource-tai, 2023-09-08:
@cumtliang 
We have collected your requirements.

cumtliang, 2023-09-08:
谢谢，
请问是否可以也增加一个下来框，显示每一个快照的编号，每次收到一个快照，就更新这个下拉框的编号。如果用户点击下拉框，立即停止采集，如果用户选中某个编号，就加载这个快照的波形。如果再次开始采集，整个环形缓冲区被清空，下拉框也被清空，当前编号显示0.

 <https://user-images.githubusercontent.com/14148856/266483637-cb11d4fb-0a16-4c7d-bb8a-8b40f83df5c3.png>
每次启动采集之后，从示波器读取的数据然后绘制曲线。在程序中开辟一个4GB的环形缓冲存储区，每次从DsoSnapshot 对象的的快照就存储到这个4GB的环形缓冲区，快照的最大数量，取决于快照的大小。程序要自动计算出最大的快照数量。环形缓冲区满了之后，就把旧的快照覆盖。停止采集后，可以把快照从环形缓冲区中把数据重新载入。