DreamSourceLab/DSView #671: v1.3.0 don't work on macos (The received packet format is incorrect!)

iopq1, 2023-07-28:
hello. this update v1.3.0 don't work on macbook air m1 v13.3.1 
The received packet format is incorrect!

dreamsource-tai, 2023-07-28:
@iopq1 
Have you used the USB hub? Please connect the device to the PC without hub.

iopq1, 2023-07-28:
yes sir. now i connected type-c to type-c but not change

dreamsource-tai, 2023-07-28:
@iopq1 
Can you try the version 1.2.2?

iopq1, 2023-07-28:
yes. i'm use 1.2.2 right now

dreamsource-tai, 2023-07-31:
@iopq1 
Does version v1.22 have this issue?

iopq1, 2023-07-31:
1.22 is works

dreamsource-tai, 2023-07-31:
＠iopq1
Please help us test v1.3 version. Can you see the waveform data when this issue occurs? Is the indicator light green? Can you make some parameter modifications to the device?

iopq1, 2023-07-31:
ok. i can't see waveform. led is green. yes i can modification the device

dreamsource-tai, 2023-07-31:
＠iopq1
Is this problem present in every acquisition? Is it possible that the device is in unstable contact?

iopq1, 2023-07-31:
1.2.2 always good work. 
https://freeimage.host/i/HZMK8bt

1.3.0 never work
https://freeimage.host/i/HZMKtm7
https://freeimage.host/i/HZMf37V

i have a reliable contact.

dreamsource-tai, 2023-07-31:
@iopq1 
Please send a screenshot of the device options dialog box for a look.

iopq1, 2023-07-31:
https://freeimage.host/i/HZMqtVe

dreamsource-tai, 2023-07-31:
@iopq1 
No, not the usb bus information. On DSView, click the "Options" button on toolbar to open the options dialog, then take a screenshot.

iopq1, 2023-07-31:
https://freeimage.host/i/HZMCk4s

dreamsource-tai, 2023-07-31:
@iopq1 
Click the button "Help"->"Log Options" to open log dialog, select the log level 5, and checked "Save to file", then caputure data one time,  and upload the log to here.

iopq1, 2023-07-31:
`DSView: Start collect.
sr: dslogic: Set device loop mode:0
sr: lib_main: Start collect.
sr: lib_main: Collect thread start.
DSView: ------->decode thread start
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
srd: Creating new usb_power_delivery instance usb_power_delivery-1.
srd: Setting channels for instance usb_power_delivery-1 with list of 1 channels.
srd: Setting channel mapping: cc1 (PD ch idx 0) = input data ch idx 0.
srd: Final channel map:
srd:  - PD ch idx 0 (cc1) = input data ch idx 0 (required)
srd:  - PD ch idx 1 (cc2) = input data ch idx -1 (optional)
DSView: decoder start sample:0, end sample:10000383, count:10000384
srd: Setting session 1 samplerate to 1000000.
srd: Registering new callback for output type OUTPUT_ANN.
srd: Calling start() of all instances in session 1.
srd: Calling start() of instance usb_power_delivery-1.
srd: Instance usb_power_delivery-1 creating new output type OUTPUT_ANN as oid 0 (usb_power_delivery-1).
srd: Instance usb_power_delivery-1 creating new output type OUTPUT_BINARY as oid 1 (usb_power_delivery-1).
srd: Instance usb_power_delivery-1 creating new output type OUTPUT_META as oid 2 (usb_power_delivery-1).
sr: dsl: Arm FPGA done
sr: dslogic: Starting acquisition.
sr: dslogic: Sending SR_DF_HEADER packet.
sr: session: Running...
sr: dsl: receive_header: trigger packet data error.
sr: dsl: 43055393604: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43055414703: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43055434428: receive_transfer(): status 0; timeout 0; received 7680 bytes.
DSView: send to decoder times: 0
srd: Freeing instance usb_power_delivery-1.
srd: usb_power_delivery-1: Resetting decoder state.
srd: usb_power_delivery-1: Releasing initial pin state.
srd: Destroyed session 1.
DSView: ------->decode thread end
sr: dsl: 43055456453: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43055475222: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: Call finish_acquisition()
sr: dsl: finish_acquisition: send SR_DF_END packet
DSView: ------------SR_DF_END packet.
sr: dslogic: remove_sources: remove fds from polling
sr: dsl: Send command:"bmFORCE_RDY"
sr: lib_main: Collect thread end.
DSView: Start collect.
sr: dslogic: Set device loop mode:0
sr: lib_main: Start collect.
sr: session: Destroy the old session.
sr: lib_main: Collect thread start.
DSView: ------->decode thread start
srd: Creating new usb_power_delivery instance usb_power_delivery-1.
srd: Setting channels for instance usb_power_delivery-1 with list of 1 channels.
srd: Setting channel mapping: cc1 (PD ch idx 0) = input data ch idx 0.
srd: Final channel map:
srd:  - PD ch idx 0 (cc1) = input data ch idx 0 (required)
srd:  - PD ch idx 1 (cc2) = input data ch idx -1 (optional)
DSView: decoder start sample:0, end sample:10000383, count:10000384
srd: Setting session 2 samplerate to 1000000.
srd: Registering new callback for output type OUTPUT_ANN.
srd: Calling start() of all instances in session 2.
srd: Calling start() of instance usb_power_delivery-1.
srd: Instance usb_power_delivery-1 creating new output type OUTPUT_ANN as oid 0 (usb_power_delivery-1).
srd: Instance usb_power_delivery-1 creating new output type OUTPUT_BINARY as oid 1 (usb_power_delivery-1).
srd: Instance usb_power_delivery-1 creating new output type OUTPUT_META as oid 2 (usb_power_delivery-1).
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
sr: dsl: Arm FPGA done
sr: dslogic: Starting acquisition.
sr: dslogic: Sending SR_DF_HEADER packet.
sr: session: Running...
sr: dsl: receive_header: trigger packet data error.
sr: dsl: 43061112219: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43061143877: receive_transfer(): status 0; timeout 0; received 7680 bytes.
[... 36 lines cut ...]
sr: dsl: Arm FPGA done
sr: dslogic: Starting acquisition.
sr: dslogic: Sending SR_DF_HEADER packet.
sr: session: Running...
DSView: Stop collect.
DSView: Data is uploading from device data buffer, waiting for stop.
sr: dsl: receive_header: trigger packet data error.
sr: dsl: 43075586879: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43075586905: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43075587083: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43075587256: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: 43075587421: receive_transfer(): status 0; timeout 0; received 7680 bytes.
sr: dsl: Call finish_acquisition()
sr: dsl: finish_acquisition: send SR_DF_END packet
DSView: ------------SR_DF_END packet.
sr: dslogic: remove_sources: remove fds from polling
sr: dsl: Send command:"bmFORCE_RDY"
sr: lib_main: Collect thread end.
DSView: send to decoder times: 0
srd: Freeing instance usb_power_delivery-1.
srd: usb_power_delivery-1: Resetting decoder state.
srd: usb_power_delivery-1: Releasing initial pin state.
srd: Destroyed session 3.
DSView: ------->decode thread end
`

dreamsource-tai, 2023-07-31:
@iopq1 
This readability is not friendly. Please upload a log with a line break. Thank you!

iopq1, 2023-07-31:
https://dropmefiles.com/D6HQI

dreamsource-tai, 2023-07-31:
@iopq1 
Do you have the conditions to test your device on Windows or Linux? Using v1.3 version.

iopq1, 2023-07-31:
i have a windows machine. wow. 1.3.0  work in real time on windows machine

dreamsource-tai, 2023-08-04:
@iopq1 
Please help us test this version．
https://www.dreamsourcelab.com/software/DSV-1.3m.dmg

iopq1, 2023-08-04:
unfortunately it doesn't change.  it's a log file https://dropmefiles.com/T4per

iopq1, 2023-08-04:
i got a macbook on intel CPU. i get same error

iopq1, 2023-08-06:
mac mini (v10.13.6 High Sierra) same error

dreamsource-tai, 2023-08-07:
@iopq1 
"The received packet format is incorrect."
Did this error message appear after clicking the stop button?

iopq1, 2023-08-07:
sometimes it was before and after clicking stop button

dreamsource-tai, 2023-08-07:
@iopq1 
Will this error also pop up automatically without clicking the stop button and waiting for a certain amount of time?

iopq1, 2023-08-07:
no. green led is blink. window shows "Triggered! -2147483648% Captured". it's continue enough long time

dreamsource-tai, 2023-08-07:
@iopq1 
Please help us to get the log from new version. You shuld clear the log first. Thank you! 
https://www.dreamsourcelab.com/software/DSV-1.3m2.dmg

iopq1, 2023-08-07:
thanks you sir. this version is work. log file here - https://dropmefiles.com/0MFkv

dreamsource-tai, 2023-08-08:
Is that issue no longer present?

iopq1, 2023-08-08:
oh no. this problem is appeared again. https://dropmefiles.com/gAN1R

iopq1, 2023-08-08:
maybe you need a macos log file - https://dropmefiles.com/h9Ltz

iopq1, 2023-08-09:
https://freeimage.host/i/HZMqtVe
pid 0x0020
vid 0x2a0e

coolbrian, 2023-08-17:
Same behavior here.  I am using Macbook Air M1, running Ventura 13.4.1.
I have tried connecting DSLogic to my Macbook without a hub and the problem ("The received packet format is incorrect!") persists.
Do you need any more data to help you locating the problem?

dreamsource-tai, 2023-08-17:
@iopq1 
@coolbrian 
I uploaded the test version, please download it.
https://www.dreamsourcelab.com/software/DSV-1.3m3.dmg

iopq1, 2023-08-18:
i have done some tests and seems that this version is worked

dreamsource-tai, 2023-08-18:
@iopq1 
Ok, thank you!

coolbrian, 2023-08-19:
@DreamSourceLab
Seems to be working now!  No more the dreaded "The received packet format is incorrect!" message!
Thank you!

AndrewCapon, 2023-09-09:
Hi Guys,

Is a signed version of this 1.3m3 going to be made available for download?

Thanks

Andy