DreamSourceLab/DSView #674: 编译报错 提示CMake Error: File config.h.in does not exist.

huming2207, 2023-08-07:
您好 @dreamsource-tai @DreamSourceLab 

刚想在mac上面试一下编译1.3.0，运行`cmake .`提示如下，似乎您忘了提交这个`DSView/config.h.in`：

```
CMake Deprecation Warning at CMakeLists.txt:22 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error: File /Users/hu/Downloads/DSView/DSView/src/config.h.in does not exist.
CMake Error at CMakeLists.txt:38 (configure_file):
  configure_file Problem configuring file

-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2")
-- Checking for one of the modules 'glib-2.0'
----- glib-2.0:
-- 	 includes:/opt/homebrew/Cellar/glib/2.76.4/include/glib-2.0/opt/homebrew/Cellar/glib/2.76.4/lib/glib-2.0/include/opt/homebrew/opt/gettext/include/opt/homebrew/Cellar/pcre2/10.42/include
-- 	 libraries:/opt/homebrew/Cellar/glib/2.76.4/lib/libglib-2.0.*
----- python3:
-- 	 includes:/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11
-- 	 libraries:/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/libpython3.11.dylib
----- FFTW:
-- 	 includes:/opt/homebrew/include
-- 	 libraries:/opt/homebrew/lib/libfftw3.dylib
----- libusb-1.0:
-- 	 includes:/opt/homebrew/include/libusb-1.0
-- 	 libraries:/opt/homebrew/lib/libusb-1.0.dylib
----- zlib:
-- 	 includes:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include
-- 	 libraries:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/lib/libz.tbd
-- Performing Test HAVE_STDATOMIC
-- Performing Test HAVE_STDATOMIC - Success
----- Qt6:
-- 	 includes:/opt/homebrew/lib/QtCore.framework/Headers/opt/homebrew/lib/QtCore.framework
-- Found OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/System/Library/Frameworks/OpenGL.framework
-- Found WrapOpenGL: TRUE
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
-- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR)
----- boost:
-- 	 includes:/opt/homebrew/include
-- Found Threads: TRUE
-- Output dir: /Users/hu/Downloads/DSView/build.dir
-- Configuring incomplete, errors occurred!
```

Jackson

dreamsource-tai, 2023-08-07:
@huming2207 
请确认一下你的分支是不是release-1.3，且拉取一下最新的代码: git pull origin release-1.3。再试一下编译，看是否还有问题。

huming2207, 2023-08-08:
> @huming2207 请确认一下你的分支是不是release-1.3，且拉取一下最新的代码: git pull origin release-1.3。再试一下编译，看是否还有问题。

刚试了下，编译通过了，明天试一下接设备好不好用，先谢谢您

closed 2023-08-08 by huming2207