DreamSourceLab/DSView #676: DSLogicPlus / Mac Os: "the device is busy, switch failed"

stephaneAG, 2023-08-10:
Hi !
Newcomer to your software, it worked once, two days ago, then now nothing and the device just has its red led always fully
Mac OS X: Catalina 10.15.4
I tried DSView v0.99, V1.12, V1.2, V1.3: in any of those, can't connect

I tried / read #332, #326, #404, nothing solves my troubles

Is there any "working fix" for that ? ( even if its overkill, ex: flashing a firmware to get back my device to "factory settings" )

thanks for the help !

ps: some logs, don't know if those would help ..

`
DSView: Store log to file: "/Users/stephaneag/Library/Application Support/DreamSourceLab/DSView/DSView.log"
DSView: Set log level: 3
DSView: ----------------- version: 1.3.0-----------------
DSView: Qt:5.12.5
DSView: 2023-08-10 03:39:42
DSView: GetAppDataDir:"/Applications/DSView_1.3.0_nonWorkingOnMacAsOf04072023.app/Contents/MacOS"
DSView: GetFirmwareDir:"/Applications/DSView_1.3.0_nonWorkingOnMacAsOf04072023.app/Contents/MacOS/res"
DSView: GetUserDataDir:"/Users/stephaneag/Library/Application Support/DreamSourceLab/DSView"
DSView: GetDecodeScriptDir:"/Applications/DSView_1.3.0_nonWorkingOnMacAsOf04072023.app/Contents/MacOS/decoders"
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found new DSLogic device count: 0
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: Hotplug thread start!
srd: Failed to load decoder .DS_Store: import by name failed:ModuleNotFoundError:No module named '.DS_Store'.
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: Protocol not exists! id:can-fd
sr: lib_main: ----------One device attached,handle:0x7fbc8911bac0
sr: lib_main: Process device attach event.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found a new device,handle:0x7fbc8911bac0
sr: dslogic: Found a DSLogic device,name:"DSLogic PLus",handle:0x7fbc8911bac0
sr: dslogic: Found new DSLogic device count: 1
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
DSView: Update device list.
DSView: Set default device.
sr: lib_main: Release current actived device. name:"Demo Device"
sr: lib_main: Start activating device.
sr: lib_main: Close the previous device, name:"Demo Device"
sr: lib_main: Activating device name: "DSLogic PLus".
sr: lib_main: To open device, name:"DSLogic PLus"
sr: dsl: Open usb device instance, handle: 0x7fbc8911bac0
sr: dsl: Opened device 0x7fbc8911bac0 on 20.2, interface 0, firmware 2.0.
sr: dsl: dsl_dev_open: Device has been disconnected.
sr: dsl: DSLogic: Closing device 0 on 20.2 interface 0.
sr: lib_main: Open device error! Will switch to "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Protocol not exists! id:can-fd
sr: lib_main: Release current actived device. name:"Demo Device"
sr: lib_main: Start activating device.
sr: lib_main: Close the previous device, name:"Demo Device"
sr: lib_main: Activating device name: "DSLogic PLus".
sr: lib_main: To open device, name:"DSLogic PLus"
sr: dsl: Open usb device instance, handle: 0x7fbc8911bac0
sr: dsl: Opened device 0x7fbc8911bac0 on 20.2, interface 0, firmware 2.0.
sr: dsl: dsl_dev_open: Device has been disconnected.
sr: dsl: DSLogic: Closing device 0 on 20.2 interface 0.
sr: lib_main: Open device error! Will switch to "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Protocol not exists! id:can-fd
DSView: Update device list.
sr: lib_main: Release current actived device. name:"Demo Device"
sr: lib_main: Start activating device.
sr: lib_main: Close the previous device, name:"Demo Device"
sr: lib_main: Activating device name: "DSLogic PLus".
sr: lib_main: To open device, name:"DSLogic PLus"
sr: dsl: Open usb device instance, handle: 0x7fbc8911bac0
sr: dsl: Opened device 0x7fbc8911bac0 on 20.2, interface 0, firmware 2.0.
sr: dsl: dsl_dev_open: Device has been disconnected.
sr: dsl: DSLogic: Closing device 0 on 20.2 interface 0.
sr: lib_main: Open device error! Will switch to "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Protocol not exists! id:can-fd
DSView: Update device list.

`

dreamsource-tai, 2023-08-10:
@stephaneAG 
How many processes did you run？A device can only be exclusively owned by one process and needs to be released to switch to a new process.

stephaneAG, 2023-08-12:
Hi !
after checking, it seems having a Virtual Box VM instance running ( even with USB "forwarding" disabled for a particular device ) prevents the device to be used for some reason ( .. )
Also, still no luck with last version on Mac OS X ( "incorrect packets .." error )
At least, I can use the "older" version ( no "Loop" mode, but better than nothing ;)

dreamsource-tai, 2023-08-12:
@stephaneAG 
Your device also shows an error prompt:“The received packet format is incorrect!”？

stephaneAG, 2023-08-12:
Hi again
Yes, when I use the latest version of the DSVIew software on Mac, it sadly does so :/
( I digged the issues a little, and since since I couldn’t find a quick fix, I opted for an older version that would work & allow me to my I2C stuff :)

Also ( sorry for asking this here :/ .. ): do you have any tutorial on how to write i2c/spi decoders for DSView ?

Thanks +

dreamsource-tai, 2023-08-13:
@stephaneAG 
We have a Chinese version of the tutorial.

stephaneAG, 2023-08-13:
@dreamsource-tai
Well, in that case, do you have a link ? I’ll see how far I can go with “auto-translate” + the actual code involved ( wouldn’t be the first time I kinda understand some code but not the comments 😌 )

dreamsource-tai, 2023-08-13:
@stephaneAG 
https://dreamsourcelab.cn/logic-analyzer/protocol_decoder_howto/

zach-c-d, 2025-02-20:
> Since google search provides this issue as a first result I will insert here solution for flatpak "Device is busy" issue.
> 
> Place following content into /etc/udev/rules.d/60-dreamsourcelab.rules:
> 
> > SUBSYSTEM=="usb", ATTRS{idVendor}=="2a0e", MODE="0666"
> 
> Restart pc or reload udev rules:
> 
> > sudo udevadm control --reload-rules && sudo udevadm trigger
> 
> I found this solution at [flathub git](https://github.com/flathub/com.dreamsourcelab.DSView).

Thanks very much! I did land here from github while troubleshooting the flatpak for `Device Busy, switch failed` on `Pop!_OS 22.04 LTS x86_64` and this fixed my issue.