DreamSourceLab/DSView #677: msys2 compile

nrzo, 2023-08-14:
I use msys2 to compile successfully, but the following error occurs when I click the start button. Is there any missing file?
Thanks for your answer!

nrzo@SHL MINGW64 ~/DSView-1.2/build.dir
$ ./DSView.exe
DSView: ----------------- version: 1.3.0-----------------
DSView: Qt:5.15.10
DSView: 2023-08-14 18:25:10
DSView: Error: can't get "System" page code
DSView: GetAppDataDir:"C:/msys64/home/yaoluo/DSView-1.2/build.dir"
DSView: Error: can't get "System" page code
DSView: GetFirmwareDir:"C:/msys64/home/yaoluo/DSView-1.2/build.dir/res"
DSView: Error: can't get "System" page code
DSView: GetUserDataDir:"C:/Users/luoyao/AppData/Roaming/DreamSourceLab/DSView"
DSView: Error: can't get "System" page code
DSView: GetDecodeScriptDir:"C:/msys64/home/yaoluo/DSView-1.2/build.dir/decoders"
DSView: Error: can't get "System" page code
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found a new device,handle:000002b5b10dc4b0
sr: dslogic: Found a DSLogic device,name:"DSLogic PLus",handle:000002b5b10dc4b0
sr: dslogic: Found new DSLogic device count: 1
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: Hotplug thread start!
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "DSLogic PLus".
sr: lib_main: To open device, name:"DSLogic PLus"
sr: dsl: Open usb device instance, handle: 000002b5b10dc4b0
sr: dsl: Opened device 000002b5b10dc4b0 on 1.39, interface 0, firmware 2.2.
sr: dsl: Security check pass!
sr: lib_main: Activating device end.
DSView: Switch to device "DSLogic PLus" done.
DSView: Error: can't get "System" page code
DSView: Load device profile: "C:/Users/luoyao/AppData/Roaming/DreamSourceLab/DSView/DSLogic0.ses.dsc"
DSView: SigSession::reload(), clear signals
DSView: Update rate list.
DSView: Update sample count list.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: The device's USB module version: 2.0
DSView: The cable's USB port version: 2.0
DSView: Start collect.
sr: dslogic: Set device loop mode:0
sr: lib_main: Start collect.
sr: lib_main: Collect thread start.
DSView: ------->decode thread start
DSView: decoder start sample:0, end sample:10000383, count:10000384
srd: Calling start() of all instances in session 1.
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
sr: dsl: Arm FPGA done
Segmentation fault

dreamsource-tai, 2023-08-15:
@nrzo 
Download the compiled program directly.
www.dreamsourcelab.com

nrzo, 2023-08-15:
@dreamsource-tai Thanks for your reply.

We need to make program-controlled instruments, and plan to make a .dll or command-line tool, which can be called by Python. We will share this code after the project is completed, do you have any suggestions?

sagasm, 2023-08-15:
Important: