DreamSourceLab/DSView #678: The decoder result of TDM audio data is wrong

nandttl, 2023-08-15:
The TDM decoder result's channel data is misaligned, channel 0's data is interpreted as channel1, and channel 1's data is interpreted as channel 2, etc. My setting is: 4channel, 16bit every sample.

[image]

dreamsource-tai, 2023-08-16:
@nandttl 
Please upload your data for us to test.

nandttl, 2023-08-16:
Hi this is the saved data, please have a check. https://cloud.189.cn/t/MZJzAvQZBvme （访问码：zst4）

nandttl, 2023-08-16:
[DSLogic PLus-la-230815-171600.zip](https://github.com/DreamSourceLab/DSView/files/12354301/DSLogic.PLus-la-230815-171600.zip)
Hi this is the saved data, please have a check.