DreamSourceLab/DSView #679: Invalid pulses in the captured signal (DSLogic Plus)

a-sv, 2023-08-15:
The problem is reproduced in self-test mode and on a real signal. I tried replacing the DSLogicPlus.bin file from version 1.3.0 with version 1.1.2 and that solved the problem.
Self-test:
[image]
Real signal:
[image]
Real signal with old DSLogicPlus.bin file:
[image]

dreamsource-tai, 2023-08-16:
@a-sv 
Can you provide us with the hardware id(vid/pid) of your DSLogic?

a-sv, 2023-08-16:
@dreamsource-tai 
My device identifier:  VID 2a0e, PID 0020

dreamsource-tai, 2023-08-16:
@a-sv 
what is your operation system?

a-sv, 2023-08-16:
@dreamsource-tai 
Gentoo Linux x86_64

dreamsource-tai, 2023-08-17:
@a-sv 
Please pull  the latest code to try, at the branch "master".

a-sv, 2023-08-18:
@dreamsource-tai 
The version in the "master" branch works fine. Problem solved.