DreamSourceLab/DSView #680: Sigrok-cli with DSLogic - how to enable RLE?

korken89, 2023-08-23:
Hi,

I'm using RLE to capture signals but need to move the app to use CLI tools now to automate the process.
However I am not able to find how the enable the RLE in the Sigrok CLI.

Could you advice on how to do this? Thanks!

dreamsource-tai, 2023-08-24:
@korken89 
I don't fully understand what you mean.
You are used the DSView to capture signals with RLE option, then exit the app,  and use the CLI tools.
Is that so?
What is the CLI tools?

korken89, 2023-08-24:
Hi,

The tool is sigrok-cli, I found here in the issues that dsview does not have a CLI mode and sigrok-cli was the recommended tool to use for CLI use.
However there I'm not able to find the RLE option.

Does that clear it up?

dreamsource-tai, 2023-08-25:
@korken89 
DSView have no CLI tools, but it supports RLE. 
The option at the device options dialog.

korken89, 2023-08-25:
Hi, I think you are misunderstanding my question.
The only known supported CLI is `sigrok-cli` as per this issue #382, how do I enable RLE in `sigrok-cli`?

DreamSourceLab, 2023-08-29:
@korken89 
sigrok-cli is a third-party tool, and we are not sure if it supports setting RLE. 
You can ask the Sigrok community for more information.