DreamSourceLab/DSView #685: I modified the size of the real-time memory that the oscilloscope can use.

cumtliang, 2023-09-01:
https://github.com/cumtliang/DSView-Unlock

This is a copy of DSView, where I have modified the size of the real-time memory that the oscilloscope can use. This is very helpful for capturing high-frequency signals; you can freely choose from 2M to 64M. 

I also modified the lock of the sampling rate and the time base, which is also very helpful for the flexibility of sampling. 

There are still some issues, such as compatibility issues between some sampling rates and the size of the real-time memory. It is recommended to use 24M or 12M.

juf00th, 2023-09-03:
@cumtliang Very interesting. Can you tell me where I can get more information?

dreamsource-tai, 2023-09-08:
@cumtliang 
Okay, we will refer to your implementation.

cumtliang, 2023-09-08:
是否可以也增加一个下来框，显示每一个快照的编号，每次收到一个快照，就更新这个下拉框的编号。如果用户点击下拉框，立即停止采集，如果用户选中某个编号，就加载这个快照的波形。如果再次开始采集，整个环形缓冲区被清空，下拉框也被清空，当前编号显示0.

[image]

每次启动采集之后，从示波器读取的数据然后绘制曲线。在程序中开辟一个4GB的环形缓冲存储区，每次从DsoSnapshot 对象的的快照就存储到这个4GB的环形缓冲区，快照的最大数量，取决于快照的大小。程序要自动计算出最大的快照数量。环形缓冲区满了之后，就把旧的快照覆盖。停止采集后，可以把快照从环形缓冲区中把数据重新载入。

dreamsource-tai, 2023-09-11:
@cumtliang 
 这个需求我们先记下来，然后对它进行评估，再提到开发计划里。