DreamSourceLab/DSView #687: 有没有可能示波器和逻辑分析仪联动

shunxingzhe, 2023-09-10:
我用的示波器和逻辑分析仪都是DSVIEW,有时候需要在有波形的同时有逻辑分析仪在一起显示,这个有可能吗

dreamsource-tai, 2023-09-11:
@shunxingzhe 
新产品在规划这个功能呢。

iciosakbrkj3158, 2023-10-12:
> @shunxingzhe 新产品在规划这个功能呢。

啥时候能出来啊 
已经在看货了 你要是整的快  我跟老板说一下 等等你们的东西

iciosakbrkj3158, 2023-10-12:
能不能整个设备 数模一体的 比如 8ch数字+2ch 模拟

shunxingzhe, 2024-02-26:
> 像我有兩部USB示波器(U3P100、C20P)+一部邏輯分析儀 就很可惜無法同步在一個畫面上 如果可以就太棒了 這樣擴展性才足夠

是的啊,别说两台示波器了,就是一台示波器和逻辑分析仪结合在一起就领先行业了,不过我在想参加肯定不会这样想,可能会想再出一台设备结合了示波器和逻分,让我们再掏钱

m0923678421, 2024-02-26:
> > 像我有兩部USB示波器(U3P100、C20P)+一部邏輯分析儀 就很可惜無法同步在一個畫面上 如果可以就太棒了 這樣擴展性才足夠
> 
> 是的啊,别说两台示波器了,就是一台示波器和逻辑分析仪结合在一起就领先行业了,不过我在想参加肯定不会这样想,可能会想再出一台设备结合了示波器和逻分,让我们再掏钱
掏不掏錢無所謂，工作上，能方便除錯比較重要，如果示波器多台可以同步顯示，我會買更多台U3P100，而不是單純使用示波器與邏輯分析儀合併的

另外軟體還有的缺點，就是示波器的下方量測，是整個擷取的訊號資料，而不是單純就顯示的資料作即時運算，這個部分就不太好用