DreamSourceLab/DSView #689: macos x dmg no mountable file systems

cire831, 2023-09-22:
trying to install 1.3.0 on Mac OS X Big Sur 11.7.5.  When I open the DMG downloaded it only says no mountable file systems.

I also tried running a previous version (I think 1.1.0) and it just hangs.

help.

dreamsource-tai, 2023-09-22:
@cire831 
You can try to compile the project yourself.

cire831, 2023-09-22:
yeah i thought about that.

bit of a heavy lift if a simple work around is out there.   And the DMG
should mount.  Something is weird with that and wanted to let you know
about it.

i’m also going to try my ubuntu box

any other suggestions?

Thanks.

eric

dreamsource-tai, 2023-09-25:
@cire831 
Can you take a screenshot of your error message and upload it?

cire831, 2023-09-30:
just tried to duplicate the failure with out success.

let's leave it open for a bit and see if I can replicate.   I'll close the issue if no progress in a week or so.

closed 2023-10-02 by cire831