DreamSourceLab/DSView #691: Feature Request: Ability to add notes into capture

Mathias-Buder-Bose, 2023-09-26:
I'm working with large/long signal captures and it would be nice to have to ability to add notes to signal(s) at a specific point in time. Those comments could appear in a list that once should able to search , ....

dreamsource-tai, 2023-09-27:
@Mathias-Buder-Bose 
Can using a cursor solve your problem? DbClick the time bar to create a cursor.

Mathias-Buder-Bose, 2023-09-27:
@dreamsource-tai thanks for the quick reply 😃. The curser is a good tool to mark a spot in time across all signals. However, I would live to have the ability to select e.g. a region within one or more signals and add a comment to it. All comments should be listed in a list view so I can search for a comment and jump to the linked region by e.g. clicking on it.