DreamSourceLab/DSView #692: Add custom decoders

maciekish, 2023-10-06:
Hi, i made a sigrok decoder for DCS-BIOS RS485 protocol and would like to use it in DSView as i like the interface better. What's the easiest way to add it? If i add the folder inside the application on macOS, it complains its damaged. I could re-sign the binary, but it feels like there should be an easier way?

dreamsource-tai, 2023-10-07:
@maciekish 
The difference between DSView and sigrok is that when calling self. wait(), it returns a tuple, while sigrok returns a list.

dreamsource-tai, 2023-10-07:
@maciekish 
This decoder is a sub decoder, so it can use without edit. But it miss the "tag" attribute.
You can see the error message from log.  You need append line as:

class Decoder(srd.Decoder):
    api_version = 3
    id = 'dcs_bios'
    name = 'DCS-BIOS'
    longname = 'DCS-BIOS Protocol'
    desc = 'Decoder for the DCS-BIOS protocol over UART'
    license = 'gplv2+'
    inputs = ['uart']
    outputs = []
    tags = ['bios']

maciekish, 2023-10-07:
Where do i place the file to make DSView use it without editing the app on macOS?

dreamsource-tai, 2023-10-07:
@maciekish 
On macOS, if you open DSView from the dmg file, you need to select the DSView icon, press the right button with the menu "Show  Package Contents",  find the "decoders" derectory, copy your new decoder files to here.

dreamsource-tai, 2023-10-07:
@maciekish 
Maybe you need to copy the DSView app to other direcotry first.

maciekish, 2023-10-07:
Is it possible for you to consider reading additional decoders from another directory than the app? If users without macOS development knowledge modify the app, the just get a cryptic message saying the app is damaged and needs to be deleted. I know i can remove the signature and use the app anyway, but this is not a good solution. Perhaps you could use ~/Library/DSView/decoders or ~/.dsview/decoders or something like that in the macOS version?

maciekish, 2023-10-07:
@dreamsource-tai Hi again, i tried adding the tag but it still doesnt appear in the list. This is the updated file which still works in PulseView. I'm not using .wait() anywhere and the tag is added. The file is saved to `/Applications/DSView.app/Contents/MacOS/decoders/dcs-bios` and i removed the signature to allow it to still launch.

```import sigrokdecode as srd

class Decoder(srd.Decoder):
    api_version = 3
    id = 'dcs_bios'
    name = 'DCS-BIOS'
    longname = 'DCS-BIOS Protocol'
    desc = 'Decoder for the DCS-BIOS protocol over UART'
    license = 'gplv2+'
    tags = ['DCS']
    inputs = ['uart']
    outputs = []
    annotations = (
        ('address', 'Address'),
        ('msgtype', 'Message Type'),
        ('datalength', 'Data Length'),
        ('data', 'Data'),
        ('checksum', 'Checksum'),
        ('state', 'RS485 State'),
        ('gap', 'Gap')
    )
    annotation_rows = (
        ('fields', 'Fields', (0, 1, 2, 3, 4)),
        ('state', 'RS485 State', (5,)),
        ('idle', 'RS485 Idle', (6,)),
    )
    
    def __init__(self):
        self.reset()

    def reset(self):
        self.new_message()
        self.state = 'UNINITIALIZED'
        self.last_sample_end = 0  # Initialize this variable

    def new_message(self):
        self.state = 'SYNC'
        self.byte_count = 0
        self.data_length = 0
        self.msgtype = 0
        self.address = 0

    def start(self):
        self.out_ann = self.register(srd.OUTPUT_ANN)

    def decode(self, ss, es, data):
        ptype, rxtx, pdata = data

        if ptype != 'DATA':
            return

        byte = pdata[0]
        
        # Calculate time gap in microseconds
        gap = (ss - self.last_sample_end)

        # Check for a 500us+ pause to move to SYNC state
        if gap > 50:
            self.new_message()

        if byte == 0xFC:
            self.new_message()
            return
        
        if gap > 50 and gap < 5000:
            self.put(ss-gap, ss, self.out_ann, [5, ['SYNC']])
            self.put(ss-gap, ss, self.out_ann, [6, ['{}us'.format(gap)]])

        if byte == 0xFC:
            self.put(ss, es, self.out_ann, [5, ['SYNC']])
        
        # Update the last sample end time
        self.last_sample_end = es

        if self.state == 'SYNC':
            self.state = 'RX_WAIT_ADDRESS'
            self.address = byte
            self.put(ss, es, self.out_ann, [0, ['{}'.format(self.address)]])
            self.put(ss, es, self.out_ann, [5, ['RX_WAIT_ADDRESS']])
            return

        if self.state == 'RX_WAIT_ADDRESS':
            self.state = 'RX_WAIT_MSGTYPE'
            self.msgtype = byte
            self.put(ss, es, self.out_ann, [1, ['{}'.format(self.msgtype)]])
            self.put(ss, es, self.out_ann, [5, ['RX_WAIT_MSGTYPE']])
            return

        if self.state == 'RX_WAIT_MSGTYPE':
            self.state = 'RX_WAIT_DATALENGTH'
            self.data_length = byte
            self.put(ss, es, self.out_ann, [2, ['{}'.format(self.data_length)]])
            self.put(ss, es, self.out_ann, [5, ['RX_WAIT_DATALENGTH']])
            return

        if self.state == 'RX_WAIT_DATALENGTH':
            if self.byte_count == 0:
                self.data_start = ss

            if self.byte_count < self.data_length:
                char = chr(byte)

                if char.isprintable():
                    formatted_byte = char  # Show as ASCII if printable
                else:
                    formatted_byte = '[{:02X}]'.format(byte)  # Show as Hex if not printable

                self.put(ss, es, self.out_ann, [3, [formatted_byte]])

            self.byte_count += 1

            if self.byte_count == self.data_length:
                self.state = 'RX_WAIT_CHECKSUM'
                self.put(self.data_start, es, self.out_ann, [5, ['RX_WAIT_DATA']])
            
            return

        if self.state == 'RX_WAIT_CHECKSUM':
            self.put(ss, es, self.out_ann, [4, ['{}'.format(byte)]])
            self.put(ss, es, self.out_ann, [5, ['RX_WAIT_CHECKSUM']])
            return
```

dreamsource-tai, 2023-10-08:
@maciekish 
Another way is to compile the project yourself, so the program is not an app package and there will be no signature restrictions.
Decoder execution error, relevant information can be seen through logs. You can run the program on the terminal to view the logs, or you can use the log options to view the log content.
The upper layer protocol does not need to call self. wait(), and its data is obtained by passing parameters.

jeremyherbert, 2023-10-19:
@dreamsource-tai I agree with @maciekish - having an extra directory that you can load decoders (in the home directory for example) from is probably the correct solution. If you edit the application itself, it can be hard to share decoders with other people, and also tricky to upgrade the software. You could also store your custom decoders in a git repository or something like that.