DreamSourceLab/DSView #693: Problem About U3: 每天早上开机红灯会一直闪烁,windows系统启动好后仍然会一直闪烁.只有手动拔插才会变成红灯常亮

reliXQ, 2023-10-07:
前段时间由于调试SPI需要的速度快 让公司买了你们的U3产品.  但是用起来有一点点小问题:
问题1:U2的时候 每次电脑开机也会红灯闪烁,但是等进入系统后红灯会常亮 并且打开软件可以被正常识别到,
现在这个U3,每天打开电脑 红灯常亮->红灯闪烁-> 进入系统后 红灯依旧闪烁  打开软件也不能让它正常工作,必须拔掉USB口重新插才会恢复正常.
向我们嵌入式开发的工程师 一般会用7~9个USB口, 要是谁家的东西每次开机都得拔插一下,可有够忙活的了.每天早上都得拔插下比较麻烦.建议修复下.否则就得忍受它一直不停的闪烁,让人心慌.

问题2:还有就是发热问题.U3的发热是有点大.我如果只打开DSView 但并没有让软件开始采集.这时候貌似U3就已经工作了. 这导致了有时我采集完数据后DSView软件没有关闭.一直处于最小化状态.一两个小时过去了.你会发现U3已经变成一个暖宝宝了.
建议优化下低功耗逻辑.毕竟设备的使用寿命也是很重要的一环.

dreamsource-tai, 2023-10-08:
@reliXQ 
重启后USB需要重拔的问题，我们将优化。谢谢你向我们反馈问题。

reliXQ, 2023-10-11:
下一版FW的发布会优化 USB3.0接口的初始化动作?

reliXQ, 2023-10-12:
[image]
我打开一个以前保存好的文件想查看,总是弹出这个对话框.

dreamsource-tai, 2023-10-13:
@reliXQ 
请联系客服，然后提供向日葵号，我们的人远程看一下。

reliXQ, 2023-10-17:
由于是公司电脑不方便远程, 我已经把相关视频发这个邮箱了 support@dreamsourcelab.com
现象是这样产生的,第一个窗口正常打开软件.  然后双击打开以前保存过的.dsl文件, 每打开一个就会弹一次窗.

reliXQ, 2025-02-20:
现在每天重新拔插U3已经是常态了, 我都习惯了 每天拔插一次TypeC口