DreamSourceLab/DSView #699: Enhancement of Maximizing Window

xqin93, 2023-10-22:
Typically, a maximized window won't cover the taskbar, but DSView does not. It takes the whole screen like a full screen rather a maximizing. Can we consider changing this behavior?

dreamsource-tai, 2023-10-23:
@xqin93 
You are the only person who responds to this issue, and we have not encountered this issue on our end.
What is your operation system and DSView version?

xqin93, 2023-10-23:
Hello, Tai. I am using Win10 Pro with DSview 1.3.1. I found the difference. I have two screens, DSView works well on the main monitor but always covers the taskbar on the second monitor.

dreamsource-tai, 2023-10-26:
@xqin93 
This issue can occur when there are two monitors. It is recommended to start the software on the main monitor.

daifel, 2023-12-03:
@dreamsource-tai 
I have this problem too and it's really annoying. I always use the software on the second monitor. Starting the software on the main monitor is not really a solution.

daifel, 2024-07-16:
Problem is fixed with DSView 1.3.2, thank you.