DreamSourceLab/DSView #701: DSView 1.3.1 tells me that I need to connect to a USB3 port when it is already connected to a USB3 port

croz-tech, 2023-10-22:
I am using a DSLogic U32Pro and have tried this with both Windows 10 and Mac OSX.  In both cases it will only enumerate on certain USB ports and tells me consistently I need to connect to a USB3 port for better performance (even though I am connected to a USB3 port).

dreamsource-tai, 2023-10-23:
@croz-tech 
1.Please check if your cable is connected properly;
2.Please use a 3.0 data cable;
If your device is 3.0, its accompanying data cable is also 3.0. Please confirm.

croz-tech, 2023-10-23:
Yes, the cable is firmly connected properly.  It's definitely a USB3.0 device and have tried two different USB 3.0 data cables with the same effect.  Also, sometimes the U3Pro32 sometimes stops working, disconnects from the DSView software and gives a flashing light for what seems to be no reason.  Is there any firmware upgrade available that might solve this?