DreamSourceLab/DSView #702: Unable to select decoder from list

jakeogh, 2023-10-24:
Using DSView 1.3.1, when I click the magnifying glass to list the available decoders, the decoder list window appears near the middle of the window, and the instant I move the mouse off the magnifying glass, the window disappears, so I am unable to select a decoder to add. If I instead attempt to use the keyboard, I can type into the pop-up window's search field, but no combination of arrow keys and ENTER allow me to select a decoder.

I'm on linux, using the bspwm tiling window manager.
[image]

The mouse cursor does not show up in the screenshot, but it is over the magnifying glass on the right side of the window.

dreamsource-tai, 2023-10-25:
@jakeogh 
You can try the KDE desktop system.

jakeogh, 2024-07-14:
It works fine with a non-tiling WM, but I really hope this can be fixed, just making that dialogue box allow keyboard selection (arrow down, press enter) would fix it.