DreamSourceLab/DSView #709: Decoder list is empty in DSView

Kyderio, 2023-11-15:
Decoder list is empty in DSView. Can anyone help to see why?

Thank you.

[image]

dreamsource-tai, 2023-11-16:
@Kyderio 
What is your operation system? Can you provide the log file for us?

Kyderio, 2023-11-16:
I am using Ubuntu 22.04, with QT version being 5.15.3. How can I provide the log file?

Kyderio, 2023-11-16:
@dreamsource-tai  tagging you in case it didn't give you notification

dreamsource-tai, 2023-11-17:
@Kyderio 
You can run DSView by console, and able to see the log. Then take a screen shot and upload it.

Kyderio, 2023-11-17:
[image]

Kyderio, 2023-11-17:
I am able to get the signals, just that decoder list is empty and I cannot decode in in Ubuntu 22.04. @dreamsource-tai

dreamsource-tai, 2023-11-20:
@Kyderio 
Is your app compiled by cmake? Please see the INSTALL file.

Kyderio, 2023-11-20:
@dreamsource-tai yes, I have followed that INSTALL file.

dreamsource-tai, 2023-11-21:
@Kyderio 
"/usr/local/share/libsigrokdecode4DSL/decoders"
Does this directory exist in your system?Copy "decoders" from DSView source directory to "/usr/local/share/libsigrokdecode4DSL".

Kyderio, 2023-11-21:
[image]
@dreamsource-tai I have the directory even before doing manual copy of decoders list.  Thank you.
[image]

dreamsource-tai, 2023-11-21:
@Kyderio 
So you still haven't seen the decoder list?
Type command on console:whereis DSView
It prints the directory of DSView app, then copy "decoders" directory to here, try again.

dreamsource-tai, 2023-11-21:
@Kyderio 
I didn't see the output information of the decoder directory from the log.

Kyderio, 2023-11-21:
[image]  I have copied the decoder directory here but there is still no decoder list..

Kyderio, 2023-11-21:
Okay, I have found it that it can work now, the application must be run from /usr/local/bin.. Thank you.

closed 2023-11-21 by Kyderio

tedy58, 2024-04-12:
I try install it on Debian 12 because I have compiled and installed version 1.3.0, so I compile and have same problem, but after I just after 
`make` did and
` sudo make install ` 
and now decoders appear even with the appended the one for ESMT and WB25Q128BV that I add to spiflash directory of decoders.