DreamSourceLab/DSView #714: 能否为linux发行版提供一个通用的Appimage?

PTALTS-LK, 2023-12-11:
对于部分linux用户来说,从源代码构建并安装程序是困难的

dreamsource-tai, 2023-12-12:
@PTALTS-LK 
这个比较麻烦，linux版本多，系统库版本也不同。
其实在linux下搭建编译环境很方便的。可以看一下INSTALL文件。

PTALTS-LK, 2023-12-12:
> @PTALTS-LK 这个比较麻烦，linux版本多，系统库版本也不同。 其实在linux下搭建编译环境很方便的。可以看一下INSTALL文件。

appimage里面类似于一个容器,运行时是通过容器里面的运行库来启动程序的,可以避免系统库不同导致的无法运行,而且运行就和win下的单exe启动一样简单
其实DSView的appimage已经有人搞了,只不过有点旧
https://github.com/HexRx/DSView-AppImage-docker
https://github.com/jolinux/AppImage-DSView

xyx0826, 2024-06-17:
我今天遇到了这个问题，需要把最新版的 DSView 打包成 AppImage. 我写了一个 Dockerfile 来编译并打包 DSView：[xyx0826/pulseview-dsview-docker-builder](https://github.com/xyx0826/pulseview-dsview-docker-builder)

编译依赖和步骤都最大程度参照主线 PulseView, 用了一个 patch 让 sigrok-utils 里的 AppImage 脚本支持 DSView.