DreamSourceLab/DSView #715: Error in PDI decoder: TypeError:argument 2 must be int, not None

mjk-gh, 2023-12-14:
Whenever I activate the AVR PDI decoder, I always get this error message
on top of the space where the decoded content is supposed to be printed:

```
Protocol decoder instance avr_pdi-1: :TypeError:argument 2 must be int, not None.
```

A part of my signal gets decoded (cannot say yet, if correct or not, I just
started debugging).

The probably relevant excerpt of the console log messages:

```
DSView: Switch to file "DSLogic PLus-la-231214-202317.dsl" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: SigSession::reload(), clear signals
DSView: Update rate list.
DSView: Update sample count list.
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 2484, resource id: 18885680, major code: 40 (TranslateCoords), minor code: 0
DSView: Start collect.
sr: lib_main: Start collect.
sr: lib_main: Collect thread start.
DSView: ------->decode thread start
DSView: decoder start sample:0, end sample:3952639, count:3952640
srd: Calling start() of all instances in session 1.
DSView: ------------SR_DF_END packet.
sr: lib_main: Collect thread end.
srd: Protocol decoder instance avr_pdi-1: :TypeError:argument 2 must be int, not None.
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 576, in decode
    self.handle_clk_edge(clock_pin, data_pin)
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 571, in handle_clk_edge
    self.handle_bits(bit_ss, bit_es, bit_val)
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 488, in handle_bits
    self.put(self.break_ss, self.break_es, self.out_ann,
TypeError: argument 2 must be int, not None

DSView: Failed to call srd_session_send:Protocol decoder instance avr_pdi-1: :TypeError:argument 2 must be int, not None
DSView: send to decoder times: 38
srd: Destroyed session 1.
DSView: ------->decode thread end
```

See attached files for screenshot of error message and for the AVR PDI settings dialog:

[image]
[image]

dreamsource-tai, 2023-12-15:
@mjk-gh 
Could you provide youre dsl data for me?

dreamsource-tai, 2023-12-15:
[image]

mjk-gh, 2023-12-15:
Wow, your PDI decoder output looks ... different. :) In front
of` the flag-like marker "D", it says "1-0" on your screenshot,
while mine says "0,1", so with a comma, not a dash:

[image]

Here is the requested .dsl file:

  [dslogic-pdi-100-khz-capture.zip](https://github.com/DreamSourceLab/DSView/files/13683521/dslogic-pdi-100-khz-capture.zip)

dreamsource-tai, 2023-12-18:
@mjk-gh 
[pd.zip](https://github.com/DreamSourceLab/DSView/files/13698618/pd.zip)
Replace to the file "decoders/avr_pdi/pd.py".
Please test and check if the results are correct, and then provide feedback to us. Thank you.

mjk-gh, 2023-12-18:
Thank you for this patch, it seems to work like a charm!

Every packet is now decoded, and after checking a few
packets, I would claim that they are decoded correctly.

Thanks!

mjk-gh, 2023-12-18:
Forgot to mention: I tested both a saved .dsl file (the one
I attached here earlier) as well as a stream captured live,
and both look plausible.

mjk-gh, 2023-12-18:
I don't know what to say. While further testing, I happened
to come across another bug. It seems that I cannot reproduce
the good AVR PDI decoder results with a live capture, instead
the capture aborts after a few seconds, and the log shows this:

```
---- version: 1.3.1-----------------
DSView: Qt:5.15.8
DSView: 2023-12-18 15:38:40
DSView: GetAppDataDir:"/usr/local/share/DSView"
DSView: GetFirmwareDir:"/usr/local/share/DSView/res"
DSView: GetUserDataDir:"/home/mjk/.local/share/DreamSourceLab/DSView"
DSView: GetDecodeScriptDir:"/usr/local/share/libsigrokdecode4DSL/decoders"
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found a new device,handle:0x562ad75b72e0
sr: dslogic: Found a DSLogic device,name:"DSLogic PLus",handle:0x562ad75b72e0
sr: dslogic: Found new DSLogic device count: 1
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: ----------One device attached,handle:0x562ad75b72e0
sr: lib_main: Hotplug thread start!
sr: lib_main: Process device attach event.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found new DSLogic device count: 0
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "DSLogic PLus".
sr: lib_main: To open device, name:"DSLogic PLus"
sr: dsl: Open usb device instance, handle: 0x562ad75b72e0
sr: dsl: Opened device 0x562ad75b72e0 on 3.6, interface 0, firmware 2.0.
sr: lib_main: Activating device end.
DSView: Switch to device "DSLogic PLus" done.
DSView: Load device profile: "/home/mjk/.local/share/DreamSourceLab/DSView/DSLogic0.ses.dsc"
DSView: SigSession::reload(), clear signals
DSView: Update rate list.
DSView: Update sample count list.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
DSView: The device's USB module version: 2.0
DSView: The cable's USB port version: 2.0
DSView: Start collect.
sr: dslogic: Set device loop mode:0
sr: lib_main: Start collect.
sr: lib_main: Collect thread start.
DSView: ------->decode thread start
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
DSView: decoder start sample:0, end sample:60000255, count:60000256
srd: Calling start() of all instances in session 1.
sr: dsl: Arm FPGA done
sr: dsl: 12679288396: receive_trigger_pos(): status 0; timeout 0; received 512 bytes.
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: Protocol decoder instance avr_pdi-1: :MemoryError:.
srd: Traceback (most recent call last):
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 585, in decode
    self.handle_clk_edge(clock_pin, data_pin)
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 580, in handle_clk_edge
    self.handle_bits(bit_ss, bit_es, bit_val)
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 555, in handle_bits
    self.handle_byte(byte_ss, byte_es, data_val)
  File "/usr/local/share/libsigrokdecode4DSL/decoders/avr_pdi/pd.py", line 290, in handle_byte
    self.insn_wr_counts.extend(self.insn_rep_count * [width_data])
[... 68 lines cut ...]
DSView: decoder start sample:0, end sample:60000255, count:60000256
sr: dslogic: dev_acquisition_start: Stop Previous DSLogic acquisition!
srd: Calling start() of all instances in session 1.
sr: dsl: Arm FPGA done
sr: dsl: 13956418637: receive_trigger_pos(): status 0; timeout 0; received 512 bytes.
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
srd: self.cmd_ss is null
```

I'll try to narrow down this bug and report back.

mjk-gh, 2023-12-18:
Here are 5 more cases:

- Case "4 long": a longer (more than 30 seconds) PDI transfer, where DSView crashed about 5 seconds after capture start and left this logfile:
  - [4long.txt](https://github.com/DreamSourceLab/DSView/files/13706117/4long.txt)
- Case "5 short": a short PDI transfer, where DSView showed the memory error message about 1-2 seconds after capture start, decoding stops somewhere in the PDI transfer:
  - [5short.txt](https://github.com/DreamSourceLab/DSView/files/13706140/5short.txt)
  - [5short.dsl.zip](https://github.com/DreamSourceLab/DSView/files/13706146/5short.dsl.zip)
  - [image]
- Case "6 short": Same as case "5 short", but this time, in buffer mode (instead of stream mode). After the buffer-mode capture ends, the captured data is displayed, the decoder says "zoom in for details", and after a few seconds, DSView crashes and leaves this logfile:
  - [6short.txt](https://github.com/DreamSourceLab/DSView/files/13706275/6short.txt)
- Case "7 short": Same as case "6 short", but this time, I set the capture to 20 seconds, because with the 5 seconds settings, DSView crashed after a few seconds after the end of the capture, so I did not have enough time for both screenshot and .dsl file. Interestingly, with 20 seconds capture length, DSView did not crash at all, so I had enough time to collect some data:
  - [7short.txt](https://github.com/DreamSourceLab/DSView/files/13706347/7short.txt)
  - [7short.dsl.zip](https://github.com/DreamSourceLab/DSView/files/13706350/7short.dsl.zip)
  - [image]
- Case "8 short": Same as case "7 short", but this time, even at the same 20 seconds capture length, DSView crashed within a few seconds after the end of the capture, so only a log file survives:
  - [8short.txt](https://github.com/DreamSourceLab/DSView/files/13706406/8short.txt)

mjk-gh, 2023-12-20:
And another little piece of information: These crashes and/or memory error messages do not happen, if the AVR PDI decoder is disabled.

dreamsource-tai, 2023-12-21:
@mjk-gh 
We need a little time to fix this issue.

mjk-gh, 2023-12-21:
No problem, thank you for the heads-up!

dreamsource-tai, 2023-12-27:
@mjk-gh 
[avr-pdi.zip](https://github.com/DreamSourceLab/DSView/files/13774846/avr-pdi.zip)
Please verify if the decoding result is correct and provide feedback to us.