DreamSourceLab/DSView #717: Can't select decoding format+ cant get decoding to work

yardenmodan, 2023-12-27:
Hey, I got recently DSLogic Basic, and it seem's I cant use the  decoding feature nor selecting the required protocol. I would appreciate your help. 
[image]

dreamsource-tai, 2023-12-28:
@yardenmodan 
Your wave data have no edge, so the spi can't work well.

dreamsource-tai, 2023-12-28:
@yardenmodan 
You mean that you can't open the decoder options dialog? click the left icon to open it.

yardenmodan, 2023-12-28:
> @yardenmodan You mean that you can't open the decoder options dialog? click the left icon to open it.
No. I mean I can't open list of decoding options like ascii hex binary, etc. it's disabled. As well as the decoding feature which can't be displayed for some reason. Regarding wave has no edge, how to fix it?

dreamsource-tai, 2023-12-28:
@dreamsource-tai 
[image]
You mean the format list can't select like this picture I uploaded?
If your signal is all 0 or all 1, it is a no edge data, spi can't decode it. Maybe you need to check if the device parameters are set correctly on options dialog, for example, the voltage.

yardenmodan, 2023-12-28:
My signal is not all 0 or 1, it does change. I tried changing the voltage threshold, it doesn't work. Can you give clear insturctions?

dreamsource-tai, 2023-12-29:
@yardenmodan 
Please upload your data to here, ensure that the decoder has been added and set up. We will reproduce your issue and analyze it. Thank you!

yardenmodan, 2023-12-30:
the problem comes from sampling rate, as long as sampling rate too low, decoding cannot be done since there is no interpretation for the data in miso and mosi. solution: increase sampling rate