DreamSourceLab/DSView #727: DSView hangs on the splash screen (Waiting for trigger! 0% Captured) and does not work

RWTLAB, 2024-01-21:
[image]

dreamsource-tai, 2024-01-22:
＠RWTLAB
Are you connecting your device through a USB hub?
Please connect the device directly to the host and try again.

RWTLAB, 2024-01-22:
> ＠RWTLAB Are you connecting your device through a USB hub? Please connect the device directly to the host and try again.

Hello, I connected the device directly to the computer and am testing on two different computers with the latest software and got the same result.

RWTLAB, 2024-01-22:
Maybe there are some problems at the hardware level?

Kabron287, 2024-11-15:
The same problem:
**DSLogic U2Dasic, DSView 1.3.2**
**Waiting for trigger! 0% Captured**
in all Operation Modes, even in **Internal Test**

RWSergo, 2024-11-15:
> The same problem: **DSLogic U2Dasic, DSView 1.3.2** **Waiting for trigger! 0% Captured** in all Operation Modes, even in **Internal Test**

If your trigger is not activated in the software, most likely your FPGA on the board has burned out. I tried to solve this problem in different ways and it did not work, I ordered a new analyzer...

Kabron287, 2024-11-15:
I think so.