DreamSourceLab/DSView #728: Decoders not running continously

c0d3z3r0, 2024-01-21:
I would expect a decoder (e.g. I2C) to run continuosly during capturing. However, decoding only takes place when either a) stopping the capture or b) hit the decoder settings button (not changing anything!). 

[image]

dreamsource-tai, 2024-01-22:
@c0d3z3r0 
What is your DSView version?
The v1.3.1 version, on stream mode, the decoder works during capturing.

c0d3z3r0, 2024-01-22:
Here is a demonstration of the problem 

https://github.com/DreamSourceLab/DSView/assets/6205797/37801f22-8a5e-419c-848e-dec77adf72ed

dreamsource-tai, 2024-01-23:
＠c0d3z3r0
From the video you provided, it can be seen that your acquisition time is 57 minutes, so the decoding progress will be slower throughout the entire acquisition process. You can set the time lower to verify that the decoder has been working continuously during the acquisition process．

c0d3z3r0, 2024-01-23:
Why does the maximum acquisition time slow down the decoding process? In my video you see that decoding happens instantly when clicking the settings button. Shouldn't the decoder be triggered each time any of the signals change?

c0d3z3r0, 2024-01-23:
Another update: I found the source of the problem. The table (protocol dock) only gets updated in 10% steps:

https://github.com/DreamSourceLab/DSView/blob/master/DSView/pv/dock/protocoldock.cpp#L480

Probably it would be better to bind update_model() to the signal decode_done. I tried to do it, but I'm not that familiar with Qt signals...

dreamsource-tai, 2024-01-24:
@c0d3z3r0 
During the collection process, if the collection is stopped midway, the decoding task can immediately end because the data volume has been determined. However, during the one hour collection process, the data volume continues to approach the total data volume, which takes an hour to complete, so the decoding progress will be slower.

c0d3z3r0, 2024-01-24:
You can clearly see in the video that it's just a view update issue. Did you even read my previous comment?

The problem is that binding decoding to the percentage update signal (each 10%) does not make any sense. That is the reason why it's "slower". However, not decoding is slower but only updating the view.

To fix this, update_model() should be bound to decode_done() instead. I just don't know how to do it bc I'm not an Qt expert.

dreamsource-tai, 2024-01-25:
@c0d3z3r0 
Sometimes, decoding generates a large amount of model data, so it is necessary to control the refresh rate.

c0d3z3r0, 2024-04-08:
Maybe it would be possible to check the amount of data and limit refresh rate based on that. This way, slow communication via  e.g. I2C can still be seen in more-or-less real-time.