DreamSourceLab/DSView #737: DSLogicPlus on DSView 0.9.9 "Hardware operation failed"

UHerschke, 2024-01-31:
Hi,
i bought the DSL+ 3 years ago and used it sometimes for small projects -> Great tool + great software! Could find my software errors on ARM & AVR mcu's.
After 1 year not using it, i got this error.
So maybe it is the same W10x64bit but some updates. It runs in vmplayer15 and additionaly i use Arduino IDE with dpinst64.
I tried to manually install driver from "C:\Program Files\DSView\Drivers\", no luck. Also tried WinUSB over Zadig software, no luck.
In the device manager it is shown "USB-based DSL instrument v2".
When i start DSView 0.9.9 unplugged it shows in the upper left corner Demo Device. If i plug in the probe it switchs to DSLogicPlus, but the probe LED stays red. On start of measuring i get this hardware error message.

dreamsource-tai, 2024-02-01:
@UHerschke 
You can update your app to v1.3.1

UHerschke, 2024-02-01:
Hi,

 

thank you for the response. I did the update yesterday evening and ..... everything is good!

 

Thank You

best regards

Ulrich

 

 
 

Gesendet: Donnerstag, 01. Februar 2024 um 01:55 Uhr