DreamSourceLab/DSView #741: UART解码- 能否给数据做个明文解码呢

reliXQ, 2024-03-07:
[image]
上面这个图片的数据解码后入下图所示:
[image]
能否在解码选项中增加一个选项 可以将数据解码成明文呢:
[image]

reliXQ, 2024-03-07:
可以做成类似下图:
[image]

reliXQ, 2024-03-14:
@dreamsource-tai   您好,不好意思刚看到消息 我说的明文是指的是:
原本HEX格式的:476F5374616E6442792E2E2E0D0A 数据其ASCII明文是:GoStandBy... 
如果能在上图接收到的HEX数据下面能把明文"GoStandBy..."显示出来.

你说的"右边面板，可以对各解码器选择显示格式的。"  具体指的哪里,可否截个图发出来,多谢多谢

reliXQ, 2024-03-26:
@dreamsource-tai 我说的这个方法可行?  还是说已经支持这个功能了 能回复下?

0x59950, 2024-04-03:
你的意思就是直接显示ASCLL码吧，如图
[image]

reliXQ, 2025-02-15:
不好意思,回复的有点晚   是的,是你上图的意思 可以将明文直接显示到数据上面?
类似这样
[image]

0x59950, 2025-02-15:
既要hex又要ascll码的话，搞两个解码器。

reliXQ, 2025-02-20:
@0x59950  请问怎么搞ascii的解码器  是哪个选项还请指教下  先谢过啦

0x59950, 2025-02-21:
再添加一个解码器