DreamSourceLab/DSView #743: 协议解析完成后，数据格式HEX模式显示问题/ After the protocol parsing is complete, the data format displays the problem/in HEX mode

1252964342, 2024-03-10:
CN:
协议解析完成后，数据格式HEX模式显示时，若协议字长大于8bit，那么起始位以及后面连续位为0且不为最后的8bit，则将不显示hex。
例如：16bit： 0000 1111 0000 1111 => FF00FF ； 12bit 0000 0000 1111 => 00FF ;

EN:
After the protocol is parsed, when the data format is displayed in HEX mode, if the protocol word is longer than 8 bits, the starting bit and the following consecutive bits are 0 and not the last 8 bits, hex is not displayed.
Such as : 16bit： 0000 1111 0000 1111 => FF00FF ； 12bit 0000 0000 1111 => 00FF ;

dreamsource-tai, 2024-03-11:
@1252964342 
这个hex就是解码器输出的结果，如果想控制长度，直接修改解码器就可以了。你用的是什么解码器？

1252964342, 2024-03-11:
您好，感谢您的回复，
目前用的最多的是SPI，通过查看0.SPI的pd.py文件内容，感觉像是:self.put(ss, es, self.out_ann, [0, ['@%02X' % self.misodata]])语句中的"%02X"得问题，不知能否更改成根据 "wordsize"大小可变%0nX？如果用if语句判断字长，可能会需要相当多的语句.....

通过更改，将0.SPI的pd.py的"%02X"改成"%04X"，是有作用的，见附件图.
[image]

而且就我自己而言，私有更改几个解码器也是挺烦的。。。。8bit，16bit，32bit都有需求

1252964342, 2024-03-11:
将set.put改成如下格式:
self.put(ss, es, self.out_ann, [0, ['@{1:0>{0}X}'.format(int((self.options['wordsize']+3)/4),self.misodata)]])
结果如下图:
[image]
经测试 8bit~32bit均可，即使为非4倍整数点。
[image]
该方法询问大佬获得。