DreamSourceLab/DSView #745: QSPI解码 command+address 解码问题

simplehu1, 2024-03-16:
DS logic  SP解码没有问题，但是QSPI解码 命令位+地址位解码的和Saleae logic不一致，从DS logic无法读取flash所发送的command及地址
[image]

Saleae logic
[image]

dreamsource-tai, 2024-03-18:
@simplehu1 
能提供测试数据吗？我们需要它来分析。

simplehu1, 2024-03-21:
[image]
[image]
[DSLogic U3Pro32-la-qspi.dsl.zip](https://github.com/DreamSourceLab/DSView/files/14681161/DSLogic.U3Pro32-la-qspi.dsl.zip)
[GD25Q256D datasheet.pdf](https://github.com/DreamSourceLab/DSView/files/14681184/GD25Q256D.datasheet.pdf)

使用DS logic解码数据如上，以及GD25Q256D flash的datasheet，可以看到设备解码没有按照QSPI解码格式进行解码，请协助

alan-Lyu, 2024-03-21:
QSPI Flash有好几种时序形式，现在的DSView里面的QSPI解码只适配了4-4-4的模式。而除了QPI，大多数情况下使用QSPI Flash都不会用到这种模式，至少不会只使用这种模式，1-4-4和1-1-4居多，还有可能混用。（楼上的时序图就是1-1-4的模式）

楼上GD的Flash也有1-4-4的模式
[image]

如果解码器可以根据命令自动切换支持的模式就更好了

附一个ISSI的Flash手册
[25LP-WP256E.pdf](https://github.com/DreamSourceLab/DSView/files/14691330/25LP-WP256E.pdf)

simplehu1, 2024-03-21:
惭愧，之前一直没注意还有那么多的模式。
之前一直用的是Saleae logic，解码没有遇到过问题，最近客户送来国产的DSView logic 让我们看看能不能用，一比较就发现这个问题了。。。国产的仍需努力啊，赶超老美

alan-Lyu, 2024-03-22:
@simplehu1 最近我抓波形的时候也遇到这个解码的问题了，一看issue还挺多人提的，感觉如果想要做到普适还有点困难，目前我能想到的解码流程就是根据cmd不同来配置不同的解码模式，不过这又涉及到不同品牌flash又有不同的命令集，decode脚本就更复杂了。

说点题外话，梦源这个逻辑分析仪对个人玩家来说确实太友好了，saleae动辄小万的价格也就公司能负担得起。硬件我用下来感觉不说有过之也无不及，软件上确实还有差距，Logic确实界面很清爽。

yunyaobaihong, 2024-04-09:
@simplehu1 
@alan-Lyu 
[pd.zip](https://github.com/DreamSourceLab/DSView/files/14914721/pd.zip)
您好，这是更新的decoder文件，替换掉软件目录下对于的decoder文件即可,更新了对于不同comand的解码功能，可以根据相关的手册进行调整

sharksmell, 2024-04-09:
我也遇到这样的问题了（QSPI 发起的cmd解码错误），对照Kingst 5016上显示的是正确的。

alan-Lyu, 2024-04-11:
@yunyaobaihong 
感谢支持，附件中的decoder文件我简单尝试了一下，十分好用，新增的命令名称解码让我可以更快的定位到特定的波形，根据命令自动识别地址长度的功能也十分方便，期待更新到主线版本。

发现了两个小问题：
一个是代码中57:58有点小笔误；
另外一个是在逻辑上，各类读命令下，在cs未拉高之前，Flash会一直输出有效的数据，这部分没有在Data Transfer那一行体现；但目前raw data很全，已经足够分析使用

附使用截图：
[image]

yunyaobaihong, 2024-04-11:
@alan-Lyu 
您好，关于您提到的这些问题，请问是否可以提供一个截图和dsl文件，我这边改进一下

yunyaobaihong, 2024-04-11:
@alan-Lyu
您好，关于您提到的这些问题，请问是否可以提供截图说明以及相应的dsl文件，我这边改进一下

alan-Lyu, 2024-04-12:
@yunyaobaihong 

读问题：
[image]
以图中读命令举例，从cs拉低开始到cs拉高，输出的都是有效数据，但是现在的Data Transfer那一行，只输出了一个字节的数据，后面就没有了；我认为最后的READ_BYTE_SINGLE一直保持到cs拉高会更好一点；

yunyaobaihong, 2024-04-15:
@alan-Lyu 
[pd.zip](https://github.com/DreamSourceLab/DSView/files/14972979/pd.zip)
您好，根据您提出的问题，新增了READ_BYTE_XX_CONTINUOUS，如果需要将接下来的所有数据都当作读数据，可以在命令列表中使用READ_BYTE_XX_CONTINUOUS替换READ_BYTE_XX。只有在CS发生跳变的时候将所有数据重置时，才能重新读取新的命令
如果仍有问题，欢迎随时反馈，谢谢

alan-Lyu, 2024-04-15:
@yunyaobaihong 
感谢更新。

关于持续读取的问题，按照我的理解，凡是有数据回读的命令都是以cs拉高为截止标志。我目前用过的一些Nor Flash型号，手册中如此描述：
Winbond - W25Q128FW
[image]
Gigadevice - GD25Q256
[image]
ISSI - IS25LP256D
[image]

从实际的波形上看，除了各类读命令，读ID命令虽然只有几位是有效数据，但是在cs拉高前，会循环输出这些数据，也符合上面手册中的描述。个人拙见供参考。

yunyaobaihong, 2024-04-15:
@alan-Lyu 好的，感谢您的解答