DreamSourceLab/DSView #748: The DSView program sees a USB 2.0 connection to a 3.0 port

harzolit, 2024-03-20:
DSView sees a USB 2.0 connection even though DSLogicPlus is connected to a USB 3.0 port (latest Windows 10 64 bit). 
I tried to update the driver for the "USB-based DSL Instrument v2" device with the 64 bit driver in the DSView library but it didn't help.
I uninstalled and reinstalled DSView and its driver after disabling driver-signature-enforcement but no change.
Help me.

dreamsource-tai, 2024-03-21:
@harzolit 
Are you using a USB hub? Is the data cable not original?

harzolit, 2024-03-21:
Hi
No, I do not use USB HUB, I use the motherboard's USB ports with the original cable of the Dslogic plus tool by DreamSourceLab.

thx

dreamsource-tai, 2024-03-22:
@harzolit 
Sorry, it has been confirmed that the DSLogic Plus is a USB 2.0 device.

harzolit, 2024-03-22:
OK, thx
-- 
 Tutanotával küldve, élvezze a biztonságos & hirdetés nélküli e-maileket! 
 https://tuta.com