DreamSourceLab/DSView #760: Assertion error in DSview

kyeakel1, 2024-04-16:
[image]
DSlogic 32pro running dsview 1.3.1, Using 19 channels with 2 parallel decoders. I have skipped (disabled) 1 channel #11

dreamsource-tai, 2024-04-17:
@kyeakel1 
Do you have to encounter this issue every time you run it? Can you provide a screenshot of the device parameters?

kyeakel1, 2024-04-17:
It happened multiple times with the setup, then the app hung and was terminated which lost the setup as I hadn’t saved it yet. The only screen shot I have is of the main screen, is that what device parameters you are referring to? If so, I will send it tomorrow as it’s on my lab computer. If not, I will set it up again tomorrow.KippSent from my iPhone

dreamsource-tai, 2024-04-17:
@kyeakel1
What I mean is the screenshot of device options dialog.

kyeakel1, 2024-04-17:
> I get the assertion fault as soon as I press start.KippSent from my iPhone

Another data point, if I enable channel 11, it works fine. So, can’t skip a channel. Sent from my iPhone

<image0.jpeg>

dreamsource-tai, 2024-04-18:
@kyeakel1 
I need to refer to your parameter settings in order to reproduce this issue. Therefore, I need you to provide screenshots of the parameters.

kyeakel1, 2024-04-18:
<74373_scan.dsc>Hope this info is sufficient to diagnose the issue.KippSent from my iPhone