DreamSourceLab/DSView #766: QSPI 解码问题，解码界面同一行 data_transfer和QSPI D0-D3对应不上

sharksmell, 2024-05-12:
逻辑分析仪：DS_Logic U2Pro16
版本：DSView 1.3.2

QSPI 解码问题，解码界面同一行 data_transfer和QSPI D0-D3对应不上
[image]
[image]

sharksmell, 2024-05-12:
另外，QSPI解码时，D0-D3显示的数据没有意义，最终数据是按照每个线路取一位来解析的，D3(bit_7) D2(bit_6) D1(bit_5) D0(bit4) D3(bit_3) D2(bit_2) D1(bit_1) D0(bit_0)

Quad SPI data input/output format:
 IO0 = (D4, D0, …..)
 IO1 = (D5, D1, …..) 
 IO2 = (D6, D2, …..)
 IO3 = (D7, D3, …..)

dreamsource-tai, 2024-05-13:
@sharksmell 
能提供一份数据给我们用来核实这个问题吗？

sharksmell, 2024-05-13:
[DSLogic U2Pro16-la-240512-101550.zip](https://github.com/DreamSourceLab/DSView/files/15288340/DSLogic.U2Pro16-la-240512-101550.zip)

DreamSourceLab, 2024-05-16:
@sharksmell 
对于第一个问题，解码的列表显示只是把每一行数据列出来，每一列之间没有时间对应关系
对于第二个问题，这个QSPI是针对QSPI flash的协议进行了改进之后的，所以会把flash相关的命令/单线/双线/四线的数据都同时显示，便于QSPI flash的调试。如果只需要单纯的QSPI四线数据解码，只要打开Q-Data这一行的显示即可
[image]