DreamSourceLab/DSView #771: Decoders not showing

planket, 2024-05-22:
When searching for decoders, nothing is populating

[image]

feedbackloopchannel, 2024-07-01:
same problem - empty list of decoders.
Ubuntu 22.04.4 LTS
DSView v1.3.2 (master at 7c53e77a13481890c4ff7c2ad39675d8253984fb)
I saw the same before when I tried to upgrade some time ago (I think it was v1.2.2).
Because of that I still have v1.2.0-RC6 installed, which works fine.

planket, 2024-07-01:
> @planket Is your DSView compiled from cmake? Please see the directory "/usr/local/share/libsigrokdecode4DSL" and run DSView from console.

This fixed my issue. Thank you.

feedbackloopchannel, 2024-07-02:
@planket could you clarify what exactly fixed your issue?
Thank you.

feedbackloopchannel, 2024-07-29:
could someone clarify this please?
the new version is unusable because of this

planket, 2024-07-29:
I followed this and it fixed the issue.

"Is your DSView compiled from cmake?
Please see the directory "/usr/local/share/libsigrokdecode4DSL" and run
DSView from console."

feedbackloopchannel, 2024-07-30:
@planket 
What does the "see the directory" mean?
I have done:
git pull
cmake .
make
./build.dir/DSView
the list of protocols is blank
I tried ./configure in libsigrokdecode4DSL dir and got this error:
error: cannot find input file: `libsigrokdecode4DSL.pc.in'

planket, 2024-07-30:
It means go to that directory and run dsview from the terminal.

feedbackloopchannel, 2024-07-30:
no, running dsview from that directory does not make any difference for me

planket, 2024-07-30:
Ah well have you tried reinstalling it? You are on Ubuntu right?

feedbackloopchannel, 2024-07-30:
Ubuntu 22.04.4 LTS
Reinstalling what? As I said, configure in that libsigrokdecode4DSL fails
I have older dsview installed, which works, and I don't want to break it.
So I am trying to run the new one from the build dir hoping to see it working before doing "make install"

planket, 2024-07-30:
I think you should make another bug report because it's different than
mine. I can't help you.

tedy58, 2024-11-19:
May be you solve the problem, but when compile and start the program from ~/DSView/build.dir$ ./DSView
I get the same issue, so for sure you have to: 
`sudo make install`
then all decoders appear normally.

gpreviato, 2025-07-10:
Hello, I have the same problem. Anyway, the problem seems that DSView search directory for decoders is misconfigured.
Decoders are installed in /usr/local/share/libsigrokdecode4DSL
But the application (in file DSView/pv/config/appconfig.cpp) looks to a different path, that coud be:
 /usr/local/share/DSView
or
../share/libsigrokdecode4DSL
In my case linking  /usr/local/share/libsigrokdecode4DSL/decoders in  /usr/local/share/DSView/decoders solve the problem.