DreamSourceLab/DSView #774: Archlinux AUR

PTALTS-LK, 2024-05-25:


dreamsource-tai, 2024-05-27:
@PTALTS-LK 
你好，我们这边用archlinux编译过，是能够编译成功的．
你的代码是最新的吗？有没有看INSTALL文件，然后安装好相应的库？
如果都没问题，那么请试一下，将文件CMakeLists.txt的671行注释，再重新编译一下．不确定是否是这个原因，等你的反馈．谢谢．

PTALTS-LK, 2024-06-01:
> @PTALTS-LK 你好，我们这边用archlinux编译过，是能够编译成功的． 你的代码是最新的吗？有没有看INSTALL文件，然后安装好相应的库？ 如果都没问题，那么请试一下，将文件CMakeLists.txt的671行注释，再重新编译一下．不确定是否是这个原因，等你的反馈．谢谢．

[image]
671行似乎不能直接注释

我使用yay(aur助手)安装dsview `yay -S dsview`

dreamsource-tai, 2024-06-03:
@PTALTS-LK 
在行的开头用#就可以注释了

dreamsource-tai, 2024-06-03:
@PTALTS-LK 
你没拉最新代码吧．

PTALTS-LK, 2024-06-05:
> @PTALTS-LK 你没拉最新代码吧．

我使用的是aur包,并不是直接从github clone的源码

PTALTS-LK, 2024-06-05:
> @PTALTS-LK 在行的开头用#就可以注释了

我直接注释了671行,得到下面输出
```
❯ cmake .
CMake Error at CMakeLists.txt:679 (endif):
  Flow control statements are not properly nested.

-- Configuring incomplete, errors occurred!
```