DreamSourceLab/DSView #777: Failing Build for DSView on Ubuntu 22.04

nmpluta, 2024-05-31:
## Issue Report: Failing Build for DSView on Ubuntu 22.04

### Description
While attempting to build DSView (SHA: 03131dd, TAG: v1.3.2) on Ubuntu 22.04 following the provided installation guide, the build fails during the `make` process. Below are the steps followed and the errors encountered.

### Environment
- **OS**: Ubuntu 22.04
- **DSView Version**: v1.3.2 (SHA: 03131dd)

### Installation Steps Followed
1. **Installed Dependencies**:
- command:
```
sudo apt install git gcc g++ make cmake libglib2.0-dev zlib1g-dev \ 
        libusb-1.0-0-dev libboost-dev libfftw3-dev python3-dev libudev-dev pkg-config
```
- output:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
libboost-dev is already the newest version (1.74.0.3ubuntu7).
libfftw3-dev is already the newest version (3.3.8-2ubuntu8).
make is already the newest version (4.3-4.1build1).
pkg-config is already the newest version (0.29.2-1ubuntu3).
git is already the newest version (1:2.34.1-1ubuntu1.11).
libglib2.0-dev is already the newest version (2.72.4-0ubuntu2.3).
libudev-dev is already the newest version (249.11-0ubuntu3.12).
libusb-1.0-0-dev is already the newest version (2:1.0.25-1ubuntu2).
python3-dev is already the newest version (3.10.6-1~22.04).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-2ubuntu9.2).
cmake is already the newest version (3.29.3-0kitware1ubuntu22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
```

- command:
```
sudo apt install qt6-base-dev libqt6svg6-dev libgl1-mesa-dev libxkbcommon-dev libvulkan-dev
```
- output:
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libvulkan-dev is already the newest version (1.3.204.1-2).
libxkbcommon-dev is already the newest version (1.4.0-1).
libqt6svg6-dev is already the newest version (6.2.4-1ubuntu1).
libgl1-mesa-dev is already the newest version (23.2.1-1ubuntu3.1~22.04.2).
qt6-base-dev is already the newest version (6.2.4+dfsg-2ubuntu1.1).
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
```
2. **Cloned DSView Source Code**:
    ```sh
    git clone https://github.com/DreamSourceLab/DSView
    cd DSView
    ```

3. **Built DSView**:
    ```sh
    cmake .
    make
    ```

### Error Log
```sh
[ 28%] Building CXX object CMakeFiles/DSView.dir/DSView/main.cpp.o
In file included from /home/npluta/DSView/DSView/main.cpp:32:
/home/npluta/DSView/DSView/pv/mainframe.h:117:10: error: ‘bool pv::MainFrame::nativeEvent(const QByteArray&, void*, long int*)’ marked ‘override’, but does not override
  117 |     bool nativeEvent(const QByteArray &eventType, void *message, long *result) override;
      |          ^~~~~~~~~~~
/home/npluta/DSView/DSView/main.cpp: In function ‘int main(int, char**)’:
/home/npluta/DSView/DSView/main.cpp:163:48: warning: ‘Qt::AA_EnableHighDpiScaling’ is deprecated: High-DPI scaling is always enabled. This attribute no longer has any effect. [-Wdeprecated-declarations]
  163 |                 QApplication::setAttribute(Qt::AA_EnableHighDpiScaling);
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/qbytearray.h:45,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qstringview.h:58,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qchar.h:666,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qstring.h:50,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qcoreapplication.h:44,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/qapplication.h:44,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/QApplication:1,
                 from /home/npluta/DSView/DSView/main.cpp:25:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qnamespace.h:473:9: note: declared here
  473 |         AA_EnableHighDpiScaling Q_DECL_ENUMERATOR_DEPRECATED_X(
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/home/npluta/DSView/DSView/main.cpp:164:40: warning: ‘Qt::AA_UseHighDpiPixmaps’ is deprecated: High-DPI pixmaps are always enabled. This attribute no longer has any effect. [-Wdeprecated-declarations]
  164 |         QApplication::setAttribute(Qt::AA_UseHighDpiPixmaps);
      |                                        ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/qt6/QtCore/qbytearray.h:45,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qstringview.h:58,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qchar.h:666,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qstring.h:50,
                 from /usr/include/x86_64-linux-gnu/qt6/QtCore/qcoreapplication.h:44,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/qapplication.h:44,
                 from /usr/include/x86_64-linux-gnu/qt6/QtWidgets/QApplication:1,
                 from /home/npluta/DSView/DSView/main.cpp:25:
/usr/include/x86_64-linux-gnu/qt6/QtCore/qnamespace.h:462:9: note: declared here
  462 |         AA_UseHighDpiPixmaps Q_DECL_ENUMERATOR_DEPRECATED_X(
      |         ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/DSView.dir/build.make:735: CMakeFiles/DSView.dir/DSView/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

### Steps to Reproduce
1. Follow the installation steps as described above.
2. Run `cmake .` and `make` in the DSView source directory.

### Expected Behavior
The build process should complete without errors, resulting in a successful compilation of DSView.

### Actual Behavior
The build fails at 28% with an error related to the `nativeEvent` function in `mainframe.h` and deprecated Qt attributes in `main.cpp`.

### Additional Information
- The `nativeEvent` function in `mainframe.h` is marked as `override` but does not override any base class method, indicating a possible issue with method signatures or inheritance.
- The warnings regarding deprecated Qt attributes suggest potential incompatibilities with Qt6.

### Suggested Solution
- Verify the correct implementation and signature of the `nativeEvent` function.
- Update the usage of deprecated Qt attributes to ensure compatibility with the latest Qt6 version.

### Request for Assistance
Please provide guidance on resolving the build errors or any necessary patches to ensure successful compilation on Ubuntu 22.04.

Thank you for your assistance.

---

**Environment Details**:
- **Qt Version**: 6.2.4
- **GCC Version**: 11.2.0
- **CMake Version**: 3.29.3
- **Boost Version**: 1.74.0
- **FFTW3 Version**: 3.3.8
- **Python Version**: 3.10.6

dreamsource-tai, 2024-06-03:
@nmpluta 
Are you using the master branch?

dreamsource-tai, 2024-06-03:
@nmpluta 
Please use the qt5, not the qt6.

Ruanxingzhi, 2024-06-05:
`sudo apt install libqt5opengl5-dev` solves the problem.

closed 2024-06-09 by nmpluta