DreamSourceLab/DSView #782: DSView v1.3.1 has no UART decoder

Semnodime, 2024-06-14:
[image]
[image]
[image]
[image]

dreamsource-tai, 2024-06-17:
@Semnodime 
Is your system linux? are you build by cmake?

Semnodime, 2024-06-17:
Maybe this issue is related to https://github.com/NixOS/nixpkgs/issues/319411. Let's wait for the PR https://github.com/NixOS/nixpkgs/pull/319641 and see.