DreamSourceLab/DSView #787: Linux install requires root/is not possible into user directory

stefanct, 2024-07-24:
Hi,
I just built DSView on Debian Bookworm and wanted to install it for myself only. This does not work because of the absolute paths within the if at `https://github.com/DreamSourceLab/DSView/blob/66ed68ea1759558082f392de1aa7643046fd39ae/CMakeLists.txt#L668`.

    cmake . -DCMAKE_INSTALL_PREFIX:PATH=...
    $ make install
    [100%] Built target DSView
    Install the project...
    -- Install configuration: "Release"
    [...]
    -- Installing: .../share/pixmaps/dsview.svg
    -- Installing: /usr/share/applications/dsview.desktop
    CMake Error at cmake_install.cmake:90 (file):
      file INSTALL cannot copy file
      ".../DSView/DSView.desktop" to
      "/usr/share/applications/dsview.desktop": Permission denied.

Changing the absolute paths to relatives ones like it is done everywhere else fixes this issue but I have not tested if a root install still works.

dreamsource-tai, 2024-07-25:
@stefanct 
sudo make install
See the INSTALL file please.

dreamsource-tai, 2024-07-25:
@stefanct 
[image]
How to use the relative path for this line? Tank you!

stefanct, 2024-07-25:
For my local install I just removed the `/` in front. As I wrote, I have not tested if that breaks a system-wide install (like the one you quoted) but I assume it works just fine. All other `install` commands use relative paths in that file after all.

dreamsource-tai, 2024-07-26:
@stefanct 
[image]
[image]

My system is ubuntu24.04. I try execute the "make insall" without sudo, but it can't work well.

stefanct, 2025-02-04:
I meant with something like this:

    diff --git a/CMakeLists.txt b/CMakeLists.txt
    index cc82669f..445210e6 100644
    --- a/CMakeLists.txt
    +++ b/CMakeLists.txt
    @@ -669,16 +669,16 @@ install(FILES DSView/icons/logo.svg DESTINATION share/icons/hicolor/scalable/app
     install(FILES DSView/icons/logo.svg DESTINATION share/pixmaps RENAME dsview.svg)
     
     if(CMAKE_SYSTEM_NAME MATCHES "Linux")	
    -	install(FILES DSView/DSView.desktop DESTINATION /usr/share/applications RENAME dsview.desktop)
    +	install(FILES DSView/DSView.desktop DESTINATION usr/share/applications RENAME dsview.desktop)
        
        add_compile_definitions(_DEFAULT_SOURCE)
     
    -	if(IS_DIRECTORY /usr/lib/udev/rules.d)
    -		install(FILES DSView/DreamSourceLab.rules DESTINATION /usr/lib/udev/rules.d RENAME 60-dreamsourcelab.rules)
    -	elseif(IS_DIRECTORY /lib/udev/rules.d)
    -		install(FILES DSView/DreamSourceLab.rules DESTINATION /lib/udev/rules.d RENAME 60-dreamsourcelab.rules)
    -	elseif(IS_DIRECTORY /etc/udev/rules.d)
    -		install(FILES DSView/DreamSourceLab.rules DESTINATION /etc/udev/rules.d RENAME 60-dreamsourcelab.rules)
    +	if(IS_DIRECTORY usr/lib/udev/rules.d)
    +		install(FILES DSView/DreamSourceLab.rules DESTINATION usr/lib/udev/rules.d RENAME 60-dreamsourcelab.rules)
    +	elseif(IS_DIRECTORY lib/udev/rules.d)
    +		install(FILES DSView/DreamSourceLab.rules DESTINATION lib/udev/rules.d RENAME 60-dreamsourcelab.rules)
    +	elseif(IS_DIRECTORY etc/udev/rules.d)
    +		install(FILES DSView/DreamSourceLab.rules DESTINATION etc/udev/rules.d RENAME 60-dreamsourcelab.rules)
        endif()
     
     endif()

If need be, one can set the prefix, cf. https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX.html#variable:CMAKE_INSTALL_PREFIX
You could even use `/` as the default to get the same behavior as before (I think) https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT.html
That way one would not have to change any files to install it without root permissions in the current directory.