DreamSourceLab/DSView #788: Decimal separator of timestamps in CSV exports depends on locale but is not quoted if it is ,

stefanct, 2024-07-24:
My Linux environment uses a rather peculiar set of locale variables to enable using English as the primary language without ridiculous other settings for dates etc. My default decimal separator is usually a comma (`,`). DSView apparently uses this for the export of timestamps as well and thus produces CSV files like this:

    head util/wrong.csv 
    ; CSV, generated by libsigrok4DSL 0.2.0 on 2024-07-24 18:29:57
    ; Channels (6/16)
    ; Sample rate: 250 MHz
    ; Sample count: 500.736 K Samples
    Time(s), SCK, MISO, 2, CS_F, MOSI, 5
    0,0,0,0,1,0,0
    4e-09,0,0,0,1,0,0
    8e-09,0,0,0,1,0,0
    1,2e-08,0,0,0,1,0,0

Note the last line where `1,2e-08` represents 12 ns. This leads to problems when further processing the file.

When executing `LC_NUMERIC="en_US.UTF-8" ./DSView` and thus forcing points (`.`) as decimal separator the export is correct:

    ; CSV, generated by libsigrok4DSL 0.2.0 on 2024-07-24 18:28:57
    ; Channels (6/16)
    ; Sample rate: 250 MHz
    ; Sample count: 500.736 K Samples
    Time(s), SCK, MISO, 2, CS_F, MOSI, 5
    0,0,0,0,1,0,0
    4e-09,0,0,0,1,0,0
    8e-09,0,0,0,1,0,0
    1.2e-08,0,0,0,1,0,0
    1.6e-08,0,0,0,1,0,0

Either DSView should just use `.` as separator or quote the respective entries with `"` if need be (or always). Pulseview works fine with my setup BTW.

dreamsource-tai, 2024-07-26:
@stefanct 
We fixed this issue, please try.

stefanct, 2025-02-04:
@dreamsource-tai 
thank you. i have tested with 2e9e2c8e726df4ef5687d39b83d4f797cc44b574 and I think it works fine.

closed 2025-02-04 by stefanct