DreamSourceLab/DSView #790: DMX512 解码错误

MosamXu, 2024-07-29:
[image]

[DSLogic U3Pro32-la-240729-190612.zip](https://github.com/user-attachments/files/16411736/DSLogic.U3Pro32-la-240729-190612.zip)

yunyaobaihong, 2024-07-30:
@MosamXu 
您好,请问是Data读取顺序错误的问题吗?
以下是修改Data读取顺序的解码器文件,希望能解决您的问题
[pd.zip](https://github.com/user-attachments/files/16420745/pd.zip)

MosamXu, 2024-07-31:
> @MosamXu 您好,请问是Data读取顺序的问题吗? 以下是修改Data读取顺序的解码器文件,希望能解决您的问题 [pd.zip](https://github.com/user-attachments/files/16420745/pd.zip)

这个错的更离谱了，连channel的值都是错的。原本只是没有加入MAB和start code的值导致channel的值整体向上错位

DreamSourceLab, 2024-07-31:
@MosamXu 右边解码结果的列表显示只是把解码结果按照字符串的方式竖向排列，不同列之间没有对应关系。这个主要是方便对解码结果的搜索和定位，可以在波形视图和列表视图之间切换。