DreamSourceLab/DSView #791: window build vcpkg qt5.15.2 minigw

ruibinlin, 2024-08-09:
x:/DSView/DSView/pv/sigsession.h:0: Note: No relevant classes found. No output generated. 会遇到好几个这个错误，网上说编码问题，编码也改了，还是不行，最终生成了exe文件，但是执行到以下步骤就crash了。

22:24:12: Starting D:\DSView\build.dir\DSView.exe...
DSView: ----------------- version: 1.3.2-----------------
DSView: Qt:5.15.14
DSView: 2024-08-09 22:24:14
DSView: GetAppDataDir:"D:/DSView/build.dir"
DSView: GetFirmwareDir:"D:/DSView/build.dir/res"
DSView: GetUserDataDir:"C:/Users/Levi/AppData/Roaming/DreamSourceLab/DSView"
DSView: GetDecodeScriptDir:"D:/DSView/build.dir/decoders"
sr: lib_main: Init libsigrok.
sr: lib_main: Scan all connected hardware device.
sr: dslogic: Scan DSLogic device...
sr: dslogic: Found new DSLogic device count: 0
sr: dscope: Scan DSCope device...
sr: dscope: Found new DSCope device count: 0
sr: backend: Register hotplug callback.
sr: lib_main: Hotplug thread start!
DSView: Set default device.
sr: lib_main: Start activating device.
sr: lib_main: Activating device name: "Demo Device".
sr: lib_main: To open device, name:"Demo Device"
sr: lib_main: Activating device end.
DSView: Switch to device "Demo Device" done.
DSView: Update device list.
DSView: Update rate list.
DSView: Update sample count list.
22:25:03: D:\DSView\build.dir\DSView.exe crashed.

请问能不能协助一下，我觉得是0: Note: No relevant classes found. No output generated.这个问题导致的 信号传递不过去
执行到这个函数就crashed了。

[image]

dreamsource-tai, 2024-08-12:
@ruibinlin 
想研究代码就用linux，windows因为编译太麻烦，很少有人在windows下编译。

ruibinlin, 2024-08-12:
@dreamsource-tai  谢谢你，已经编译出来了， 我试了两个版本1.3.1 和1.3.2  ，其中1.31.1功能完全正常，1.3.2 在mac上功能正常，在windows上 有两个问题，第一个就是
![Uploading 1723451561458.png…]
逻辑分析仪在demo状态下 载入完波形和解码后闪退了，我还在查看是什么原因。

ruibinlin, 2024-08-12:
[image]
出来图形是这样的，很缓慢，也是有一点问题。

lclclc666789, 2024-08-12:
我 整了一个visual studio 编译的版本 请参考 [****](https://github.com/lclclc666789/DSView) 的 branch:dev_msvc
亲测 visual 2022 可以编译运行，无需minigw ，直接用vs20022 就可以编译

WayneHe, 2025-01-24:
> 我 整了一个visual studio 编译的版本 请参考 [****](https://github.com/lclclc666789/DSView) 的 branch:dev_msvc 亲测 visual 2022 可以编译运行，无需minigw ，直接用vs20022 就可以编译

试了你这个，编译报错
[image]