DreamSourceLab/DSView #798: Why did the sub-MHz sampling frequencies disappear?

WinterMime, 2024-08-26:
Dear DreamSourceLab team!

I used and liked the very low frequency sampling rates for "unlimited" event logging. Why did they disappear from version v1.3.x? I can accept if most users can find it distracting or superfluous. Or has it a serious reason to remove it? I would suggest to put a checkbox in Options for example: " Enable sub MHz sampling frequencies." So it will not show by default but it is usable if somebody needs it (by ticking the checkbox).

dreamsource-tai, 2024-08-27:
@WinterMime 
Because the low sampling rate makes the waveform refresh too slow, it was removed. We will optimize it in the future. Thank you for your suggestion.

WinterMime, 2024-09-07:
I thought it was removed because the huge amount of predefined sampling frequencies are overhelming, distracting and the low frequencies are rearly used. I can admit thats all are true but I used them frequently.

I develop EtherCAT communication devices where the cycle time is 1ms/2ms/5ms and several ten kHz sample frequenty is enough. I was super fast because of the low amount of data and if I enjoyed the extreme long sampling period.

> Because the low sampling rate makes the waveform refresh too slow...

In which capture mode? Could you show me example for my better understanding?