DreamSourceLab/DSView #799: Protocol selection is blank

feedbackloopchannel, 2024-09-01:
Current master (0957a05)
Ubuntu 24.04.1 LTS (observed the same under 22.04.4 before upgrade)
cmake .
make
./build.dir/DSView

[image]
[image]

dreamsource-tai, 2024-09-02:
@feedbackloopchannel 
Have you done command : "sudo make install"

dreamsource-tai, 2024-09-02:
Install all file, and type "DSView" to run the program.

feedbackloopchannel, 2024-09-02:
No, I did not do "make install". I wanted to see it working before I replace another, much older, but working installation.
Is there a way to test it before installing?

dreamsource-tai, 2024-09-02:
@feedbackloopchannel 
If your binary file "DSView" is in the directory "bin", you need to create a directory "share" at the same level as "bin", create a sub directory "DSView" under "share", and then copy "res", "decocers" and other files from the source code directory to the "DSView" directory.

dreamsource-tai, 2024-09-02:
@feedbackloopchannel 
[image]
[image]

feedbackloopchannel, 2024-09-03:
There is no directory "decoders" in DSView (and no qt5.1 either). No "lang", but there is "languages". Are you talking about the current master branch?

dreamsource-tai, 2024-09-04:
@feedbackloopchannel 
Yes, all at the master branch.
You try the file I uploaded.
[Uploading share.zip…]()

feedbackloopchannel, 2024-09-04:
Seems like you did not finish your upload.
And why should I try some file?
I am trying to make sure I can install the current version (either from master or the latest release).

dreamsource-tai, 2024-09-04:
@feedbackloopchannel 
These files actually exist in the source code directory, and they need to be found separately and copied to the test directory to ensure that the files are complete before running DSView To save trouble, I packed them for you. You can also find these files on your own from the source code directory.

feedbackloopchannel, 2024-09-05:
you still did not upload the file. I tried to find the pieces myself and arrange something like you suggested, but DSView seems to pick up the installed old versions from /usr/local/share:
DSView: GetAppDataDir:"/usr/local/share/DSView"
DSView: GetFirmwareDir:"/usr/local/share/DSView/res"

perhaps is has to be instructed of the installation path or something.
I am starting to think that I should just do "make install" and hope it works. if it does not, hopefully I should be able to checkout the version I have from git and reinstall. I wish testing of new versions would be easier.