DreamSourceLab/DSView #805: Notes on Export

WinterMime, 2024-09-15:
Dear DreamSourceLab Team,

I would like to add some notes on File/Export function.

1. It exports now all the channels, it could be useful to provide a opportunity to select the channels to save.
2. Something is weird with file name display. I select cursors then, I change file name, but the showed file name on
 export window does not change until I hover over the mouse cursor. Sometimes the file name overlaps with 
previous filename as there are more than one label on each other.

dreamsource-tai, 2024-09-18:
@WinterMime 
[image]
the left button is the way to select the data of one decoder, and the right button is to select the data column of current decoder to export data.

dreamsource-tai, 2024-09-18:
@WinterMime 
Some system display formats can cause file paths to be garbled, but manual input can display the correct ones. If you are using Linux, please compile the latest code yourself.

dreamsource-tai, 2024-09-18:
@WinterMime 
You can try the RC version.
https://dreamsourcelab.cn/software/DSView_v1.3.3-rc3_x64_setup.exe

WinterMime, 2024-09-23:
In RC version the file name is always correct. Thank you.

closed 2024-09-23 by WinterMime