DreamSourceLab/DSView #806: Wayland support

benklop, 2024-09-15:
DSView doesn't work quite right in wayland, whether in xorg compat mode (XWayland) or natively. It seems that window positioning / translation / resize is being handled by DSView itself rather than leaving it up to the compositor / window manager, and this is causing problems.

Specifically it's not possible to move the upper left corner of the window, nor the left side or the top, any attempt to do so ends up moving the opposite edge since in wayland the app can only control its size and not its position.

It would be great if it was possible to simply disable all the custom window handling stuff and just use the default compositor / Qt window decorations.

dreamsource-tai, 2024-09-18:
@benklop 
Are you using Qt5?

benklop, 2024-09-26:
I am using Qt5, yes.

benklop, 2024-09-30:
I'm having trouble getting it to compile with Qt5 - it's failing with:

```
[ 88%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/strutil.c.o
/home/benklop/Downloads/DSView-1.3.2/libsigrok4DSL/strutil.c: In function ‘sr_parse_sizestring’:
/home/benklop/Downloads/DSView-1.3.2/libsigrok4DSL/strutil.c:343:19: error: implicit declaration of function ‘strcasecmp’; did you mean ‘g_strcasecmp’? [-Wimplicit-function-declaration]
  343 |         if (*s && strcasecmp(s, "Hz"))
      |                   ^~~~~~~~~~
      |                   g_strcasecmp
make[2]: *** [CMakeFiles/DSView.dir/build.make:3462: CMakeFiles/DSView.dir/libsigrok4DSL/strutil.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/DSView.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

I'm not sure what to make of this, or why it would only happen with Qt5 but not Qt6

dreamsource-tai, 2024-10-08:
@benklop 
What is your operation system?
You can try that replace "strcasecmp" as "strcmp".

vanbwodonk, 2024-10-15:
Hi, I'm also having a similar issue. I'm using arch linux, and installed dsview via AUR.
https://aur.archlinux.org/packages/dsview

i can only minimize and maximize window, drag window is not working.

eolder, 2024-10-24:
> I'm having trouble getting it to compile with Qt5 - it's failing with:
>
> ```
> [ 88%] Building C object CMakeFiles/DSView.dir/libsigrok4DSL/strutil.c.o [...]

FWIW I replaced these with the suggested value and it compiled and worked afterwards, I don't own the hardware so I was just testing the "demo" probe but it seemed to be working as I would have expected it to work.

benklop, 2024-10-27:
At least on Debian with a wayland desktop, compiled with Qt5 and the above suggested substitution, I get the same behavior - drag doesn't work but maximize/minimize does. That happens whether I start with `QT_QPA_PLATFORM=xcb DSView` or with `QT_QPA_PLATFORM=wayland DSView`

benklop, 2024-10-27:
I suspect this has to do with `pv/mainframe.cpp` being from 2016 - 8 years is a pretty long time in the lifespan of wayland.

ScherbakovAl, 2024-11-07:
I encountered the same error message when trying to build the program. I'm using Fedora 41.
In the end I just installed the DSView flatpak from the app store.

maks, 2024-12-16:
I built on Ubuntu 24.04 which now is Wayland be default and thanks for the top above, running with `QT_QPA_PLATFORM=xcb DSView` does allow the window to be dragged around. In terms to Qt versions I have:
```
qmake --version
QMake version 3.1
Using Qt version 5.15.13 in /usr/lib/x86_64-linux-gnu
```