DreamSourceLab/DSView #807: Feature: Loop Mode Interrupt

A1010Marcus, 2024-09-23:
I had to use the Loop Mode to debug some Hardware because of Buffer Mode Size Limitations.
There was no way to stop the Loop on a Trigger.
So i used a Developement Board to catch a Trigger emulate a Keyboard and send 's' to DSView to stop the Loop 10s after the Trigger.

DSView is already getting the Data ( live on screen ) so stoping Seconds (user setting), not instant, after a channel changes logic shuld be easy.
I used Saleale before and i'm missing this Feature.

If someone is already familiar with the Code and the change is not to time consuming. This would be a nice to have.
If not i will take a shot at it when i find the time to do so. If no one has any objections of course. 
Mind that it could take a while because i have to do it in my free time witch is limited.

dreamsource-tai, 2024-09-24:
@A1010Marcus 
Please describe clearly the functionality you hope to achieve. Do you want to automatically stop collecting after a certain period of time after triggering?

A1010Marcus, 2024-09-24:
"Do you want to automatically stop collecting after a certain period of time after triggering?"
Yes exactly. The Setting will be in seconds after a channel change in streaming mode -> loop mode

Right now i am using a U3Pro16 to record 6 Channels 5minutes before and 5minutes after a trigger without RLE Compression.
I tried RLE Compression and Buffer Mode but not enough data is gathered after the trigger.

DSView Settings are:
Operation Mode Stream Mode
Use 6 Channels
Mode: Loop
 
I use a STM32 Dev Board to send a Keyboard 's' to DSView to stop the loop 5 minutes after the trigger.
And i want to not use the STM32 Dev Board anymore.
Just DSView and the U3Pro16.

A1010Marcus, 2024-09-24:
Long story short.

I want to add a software trigger in Loop Mode that presses the stop button X seconds after a channel change.
The User settings will be on witch channel to trigger, what kind of trigger ( high, low, any ) and how many seconds to trigger the stop after a channel change is detected.

Later the trigger can be upgraded to include complex patterns like for example I2C Address.
But i want to keep IT simple at first to avoid time consuming bugs.

A1010Marcus, 2024-09-24:
As for layout. I was thinking of adding 1 Edit line in the existing Trigger - Dialog.
And gray it out or make it invisible when not in use.
Or use the existing Percentage Scroll / Edit Box and compute the required seconds delay from there.
However the reaction of the Software Trigger will be slow. I plan to avoid values under 1 Second.

The trigger Settings are already available for each channel.
U can already set them even if they don't work in Loop Mode.

dreamsource-tai, 2024-09-26:
@A1010Marcus 
The trigger in the loop model is not working. There is currently a solution to address your requirement: run in Buffer mode, set the correct trigger conditions (simple, multi-level, serial), and perform a single acquisition. When the triggering conditions are met, the hardware will start transmitting data, and when the amount of data is sufficient, it will automatically stop collecting. This is consistent with the requirement of "automatically stopping collection after a certain period of time after triggering" mentioned above.

A1010Marcus, 2024-09-26:
"The trigger in the loop model is not working. There is currently a solution to address your requirement: run in Buffer mode, set the correct trigger conditions (simple, multi-level, serial), and perform a single acquisition. When the triggering conditions are met, the hardware will start transmitting data, and when the amount of data is sufficient, it will automatically stop collecting. This is consistent with the requirement of "automatically stopping collection after a certain period of time after triggering" mentioned above.:"

That is the first thing I tried.
I read the manual and new the limitations before buying the U3Pro16.
Back then i knew it wouldn't work without some modifications.  
The amount of data and the fact that it is not simple ( does not RLE compress ) is the problem.
We are talking about 10 minutes: 5before the trigger 5after the trigger.
Buffer Mode will offer 20s without RLE compression.
I am not interested in the data near and after the trigger to determine a failure there.
Imagine a plane crash black box that does record just 20s before the crash. 
More the whole 10 minutes of data contains the signal i am interested in.
And the 5 minutes before the event are more relevant.
I wold not bother programing firmware for another development board if it was that simple.

A1010Marcus, 2024-09-26:
I forgot to mention that the minimum sample rate, required to see the signal is 10Mhz.
But i use loop mode with 20Mhz to be on the safe side.
In Buffer mode that would offer 5s without RLE compression.
In Stream mode 10minutes with 20Mhz are possible.
Only there is no way to stop the loop with the trigger in the middle of the data (5minutes before and 5minutes after)

A1010Marcus, 2024-10-02:
I tried different settings and it is possible to record 11.45 minutes at 25 Mhz in loop mode.
The U3Pro16 can stream at 500 Mhz using 6 channels.
So the interrupt in loop mode is incredibly useful. 

I am still using another Dev board to send 's' Key to DSView and stop the loop after X seconds.
However it bugs me that the window must be active.

I will try to modify the source code so the trigger and the trigger position work in loop mode.
The GUI will stay the same.
Just the Trigger Setting Simple Trigger and Trigger Position will be active in loop mode.

Any objections / suggestions?
Are u already working on the feature?

dreamsource-tai, 2024-10-08:
@A1010Marcus 
The trigger will not work in loop mode, so you can work in repeat mode. Perhaps you need a decoder to recognize that a certain collected data has met the triggering conditions.
file:sigsession.cpp, lile:572, set a timer for repeat mode at here.
When one data frame be decoded, file:data/decoderstack.h, method:annotation_callback() will be calls. at there,  you can call _session->stop_capture() to stop the collecting task.
Because we set a long timer for repeat mode, so it can got lot of data before the collecting be stoped.
May be, you can stop the timer when one data frame be decoded at sigsession.cpp.
And, in stream mode, the trigger only work as simple mode.
We currently have no plans to fulfill your request, but it may be implemented in a future version.
If you want to implement it yourself, we can provide some suggestions.